bambulab/BambuStudio #1896: No Ability To Adjust Print Order Within An Assembly

MadMax389, 2023-06-05:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Win11

### Additional system information

CPU 10th gen Core i5

### Printer

X1 Carbon Combo

### How to reproduce

In Prusa Slicer, I am able to adjust the print order within a merged group or assembly so that one object can be printed before another object. I use this ability to be able to print targeted brims using support W when the standard brim is difficult to remove. Within an assembly, Bambu Studio does not have the capability to rearrange the print order, sometimes causing the brim object to occlude the part object. To reproduce:

1. Use the uploaded .3mf or:
2. Create a 10mm cylinder primitive (part)
3. Create a cube primitive .2mm high (brim) 
4. Clone the brim, and then clone the part.
5. Create two assemblies, one with the brim first, then the part and another with the part first, then the brim.
6. Slice the plate and look at the first layer. The assembly with the brim first prints around the part (desired) while the assembly with the part first prints around the brim.
7. 
[image]

### Actual results

There is no ability to change the print order within an assembly. In Prusa Slicer, you are able to adjust the print order simply by clicking the part and dragging with the mouse. 
[image]

[image]

### Expected results

Be able to change the order of one or more parts within an assembly. The current work around is to add the parts to the plate in the proper order. This can be more complex than necessary on complicated prints.

### Project file & Debug log uploads

[Bambu Print Order Demonstration.zip](https://github.com/bambulab/BambuStudio/files/11653142/Bambu.Print.Order.Demonstration.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-08-03:
> In Prusa Slicer, you are able to adjust the print order simply by clicking the part and dragging with the mouse.

I tried these operation on prusa slicer 2.6. But I can't control the print order, but only can control the clipping order of multi part.
[image]

SaltWei, 2023-08-03:
Is there any other situation that we need to change the print order?
If only "brim ear" case, I think developing a regular brim ear function is more perfect!

MadMax389, 2023-08-03:
Perhaps "print order" is not the right term in favor of clipping order. There are other use cases where arranging the order will be beneficial. In my text example, you can see that the order of the items in the assembly affects whether the text prints or not.

[image]
[image]

I use the clipping order to make multi-layer targeted brims. In the spider example, it would be nice to be able to drag the spider to the bottom of the list. My current workaround is to copy and paste the object to create a copy at the end of the list, then delete the first one.
 
[image]
[image]
[image]

SaltWei, 2023-08-04:
If my understanding is wrong, please don't hesitate to point out.
Actually, this feature request can be devided into two features.
1 user define the clip order of multi-parts model. User can adjust the part order in the list, and the bottom one has highest priority to clip other part to remain its area. By this feature, user can add "brim part" for model and merged with model together, while keeping the whole part of original model.

2 Add one way for user to define the filament print order of first layer. So the "brim part"  which usually uses support filament can be printed first to help the attachment of original model part

MadMax389, 2023-08-04:
@SaltWei 
Yes for item 1. Just the ability to change the order of the objects in an assembly list (like Prusa Slicer) would be very convenient.

For item 2, I think just the ability to specify a different brim material (like can currently be done with support and support interface) would be a nice feature. This would allow us to specify Support W as a brim material. Also, adding the brim function as a modifier would allow targeted brims in certain areas without having to brim the whole object. (This does not exist in Prusa Slicer).

SaltWei, 2023-08-17:
Be handled in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52

closed 2023-08-17 by SaltWei