bambulab/BambuStudio #1898: Connect printer to wifi with hidden SSID

FloMiLe, 2023-06-06:
**Is your feature request related to a problem? Please describe.**
We have bought a X1C for your school, but we are unable to connect it to our devices wifi, which has a hidden SSID, and we cannot change that.

**Describe the solution you'd like**
Please add the possibility to enter an SSID on the printer's panel such that you can connect to a wifi with hidden SSID.

**Separate issue: Forum**
On an unrelated note, I have joined the forum, but cannot post new topics. Please have a look at that. Thanks.

sodachen22, 2023-06-09:
Thanks for the feedback. Yes, there is a plan for this feature.

Rhombus-Jeremy, 2023-07-07:
We have the same need. Any update on when this will be possible?

sodachen22, 2023-07-09:
Yes, this feature will be supported

compwizard7, 2023-07-31:
Due to IT Security reasons, (I don't want to talk about) we are unable to get this on our network without being able to enter an SSID. Is there an update on when this will be implemented?
Thank you.

Crowdedlight, 2023-08-10:
We are having same issue as above, can not get it on our lab network until it supports hidden ssid.

Robbbb, 2023-08-22:
Same issue, critical for our use in our workplace. 
We will not be able to buy more Bambu units until this is resolved. 
Is there a way to edit config files over SSH to hardcode an SSID?

Robbbb, 2023-08-23:
They added the feature! I was able to update the firmware by tethering/hotspot via my phone and the option to enter network name was added.

SuperDodge, 2023-08-23:
> They added the feature! I was able to update the firmware by tethering/hotspot via my phone and the option to enter network name was added.
> This issue can be marked closed AFAIK

Where did you get that option?  I just get a QR code on the screen and this when I follow the QR code.  No option to enter network name...  I am on 1.04 firmware.[](url)
[image]

Robbbb, 2023-08-24:
Directly on the machine (x1) screen.
Skip binding/registration, ignore the QR code, ignore the app.

Set your phone or laptop up as a Wi-Fi hotspot and connect to it with the
printer using Wi-Fi settings on the tiny screen. Use this temporary
internet to update the firmware on the printer. Once the firmware is
updated, go back to the Wi-Fi settings on the tiny printer screen and now
there is a join other network at the very bottom of the list of detected
Wi-Fi networks. You can enter the SSID and password manually there.

Worked for me, hope it works for everyone!

SuperDodge, 2023-08-24:
> Directly on the machine (x1) screen. Skip binding/registration, ignore the QR code, ignore the app. Set your phone or laptop up as a Wi-Fi hotspot and connect to it with the printer using Wi-Fi [...]

Thanks for the response.  On the P1P that doesn't have a touch screen your only option is to scan the QR code which opens the app and the app only lets you select visible SSIDs.  This is still an issue for P1P users.

SuperDodge, 2023-08-25:
Just in case it helps anyone else - if you made the SSID visible and connect the P1P to it, it will continue connecting to the network even if you hide the SSID again.  As my network admin I was able to do this to move forward although I understand this isn't an option for everyone.

cederom, 2023-09-25:
The same here, just got my X1C, wont see the hidden SSID, will try to update firmware and report back :-)

The best option would be Ethernet though.. surprised there is none! There is a big thread on Forums about this: https://forum.bambulab.com/t/ethernet-port/749/35

Amazing printer! Congratulations! :-)

JeremyLaurenson, 2023-10-12:
> Just in case it helps anyone else - if you made the SSID visible and connect the P1P to it, it will continue connecting to the network even if you hide the SSID again. As my network admin I was able to do this to move forward although I understand this isn't an option for everyone.

Worked for me as well.

theabstractpyro, 2024-06-06:
> Directly on the machine (x1) screen. Skip binding/registration, ignore the QR code, ignore the app. Set your phone or laptop up as a Wi-Fi hotspot and connect to it with the printer using Wi-Fi [...]

hello, I just got an A1 mini for my dorm and with the latest firmware the join other network button doesn't show up. Is there any way this will be added to the A1 mini? I don't have an SD card reader so I'm kind of SOL without wifi

nhma20, 2024-07-22:
Workaround for P1S that solved the issue for me:

1. Forget the network on the printer (which also signs out the user).
2. Change phone's hotspot SSID and password to be identical to those of the hidden network.
3. Repeat setup of printer using the phone hotspot.
4. Shut down printer.
5. Turn off hotspot.
6. Turn on printer and it should hopefully automatically connect to the hidden network now.

I originally tried to solve it by adding the wifi configuration directly to the SD card (https://wiki.bambulab.com/en/p1/manual/p1-sd-card-network-configuration-guide) but the printer would always default to the settings entered when setting up the printer through the app where it's only possible to choose visible networks.

Hope this can help others struggling with hidden networks.