bambulab/BambuStudio #1902: Static Access Code

aminoginji, 2023-06-06:
**Is your feature request related to a problem? Please describe.**
The access code changes when the device is powered off and on.

**Describe the solution you'd like**
A static access code that dosent change when powered on or off that can be set in the SD card.

**Describe alternatives you've considered**
Not turning off the machine

**Additional context**
Its a waist of power to leave the machine on when its idle

sodachen22, 2023-06-08:
The access code shall be remembered in power off/on, it is changed when the user refreshes on the screen. Which printer does this issue happen on, X1 or P1P? If the firmware is not latest, can you please have an update and test again?

sodachen22, 2023-06-09:
Yes, there is a bug fixing.

closed 2023-06-09 by sodachen22