bambulab/BambuStudio #1903: File .3mf corrupted when using "&" symbol in text fields

jmorda, 2023-06-07:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambu Labs X1C with AMS

### How to reproduce

1. New project
2. Add primitive (cube for example)
3. Add text on cube with "&" symbol ( example : Essai& )
4. Save
5. Reload -> Fail

[image]

### Actual results

Load of .3mf file fail
[image]

### Expected results

Loading the project without error

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/11680446/log.zip)
[Essai.zip](https://github.com/bambulab/BambuStudio/files/11680451/Essai.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-06-14:
Thank you for reporting this issue, it is a studio BUG, ​​we will fix it in the next version.

zhimin-zeng-bambulab, 2023-08-18:
It has been fixed in [V1.7.4](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52)

closed 2023-08-18 by zhimin-zeng-bambulab