bambulab/BambuStudio #1904: [Issue] So many issues

BambuLabDiscord, 2023-06-07:
make it work like prusa slicer. 
1. Keyboard shortcuts dont work dependably.
2. It visibility - software doesn't keep track of this correctly.  Wrong on thumbnail on the printer.
3. "Reload from disk" feature - doesn't work.  Each part doesn't remember where it was uploaded from.  It just brings up Windows Explorer to "refind' your file.

SO sick of this software.  And it being not open source means I can't automate anything.  The software is *SHIT*

lanewei120, 2023-06-08:
_Keyboard shortcuts dont work dependably._
which keyboard shortcuts?
we have added lots of shortcuts, if you think they are not enough, please give the detailed keys as suggestion
[image]

_It visibility - software doesn't keep track of this correctly. Wrong on thumbnail on the printer_
please give more details or pictures

_"Reload from disk" feature - doesn't work. Each part doesn't remember where it was uploaded from. It just brings up Windows Explorer to "refind' your file._
could you give a gif for your detailed steps? I have tried this function, it should reload it from the original place

_And it being not open source means I can't automate anything_
also we can not understand which do you mean
Bambustudio is open source, you can submit your changes to github

lanewei120, 2023-06-08:
@BambuLabDiscord 
to report a bug, please give more information to follow the bug format
it will help us to understand more and improve it

closed 2023-06-29 by lanewei120