bambulab/BambuStudio #1907: Multiple height range modifier

eddydas, 2023-06-08:
**Is your feature request related to a problem? Please describe.**
It looks like we can add only 1 heigh range modifier now. It would be great if we can add multiple height range modifiers, best if they can overlap.

**Describe the solution you'd like**
Allow us to right click an object, then "Add Height Range Modifier" and it will create new height range modifiers. And I can repeat that to add multiple height range modifier.

**Describe alternatives you've considered**
N/A

**Additional context**
I know Bambu Studio supports height range modifiers today - but it supports only 1 range. Hope you can support multiple!

liftbag, 2023-06-08:
By clicking on the + button you can add as many height range modifiers as you like.

[image]

freeformz, 2023-07-19:
What version is that?

closed 2023-07-19 by eddydas