bambulab/BambuStudio #1908: Error slicing while slice on thin walls with color

xuhao1, 2023-06-08:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Printer

Bambu X1CC

### How to reproduce

1.Import thing wall model
2. Paint it in bambu
[image]

4. Slice in normal mode or odd mode.
5. It appears to be:

### Actual results

The wrong slicing result
[image]

### Expected results

The right slicing result (but single color)
[image]

### Project file & Debug log uploads

[bug_report.3mf.zip](https://github.com/bambulab/BambuStudio/files/11689765/bug_report.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-06-13:
多色区域划分不支持航模类模型, 因为航模类模型本身有大量错误面片和非流型边导致切片出错。
[image]

xuhao1, 2023-06-14:
@SaltWei 对航模类产品，多色功能是非常重要的，很多航模产品追求外观，用户会希望能产生类似涂装的效果。
如果能够完美支持，我相信bambu会成为所有航模玩家的首选。

SaltWei, 2023-06-27:
航模类模型本身很特殊，大量的非流形面错误，mesh不是闭合的三维实体。当前的多色区域划分对错误面片是无法处理的。
我们想支持，但是这个很困难。短期内难以搞定对航模模型的处理。

xuhao1, 2023-06-30:
@SaltWei 好吧 希望你们加油

xuhao1, 2023-06-30:
@SaltWei 另一个issue是否可行？
我现在在强行用cura打航模……
https://github.com/bambulab/BambuStudio/issues/1346