bambulab/BambuStudio #1911: Feature Request: Auto color change feature for logos and signs from PrusaSlicer

sprior, 2023-06-08:
PrusaSlicer now has a feature that when you first slice an object that has something it thinks is a logo or sign on top of a flat surface it offers to automatically add a layer based color change.  While it is possible to manually add such a color change in Bambu Slicer it is a little tricky to get right at first because can be difficult to understand in the GUI on what layer to add the change and be one layer off.

With the popularity of the Bambu AMS unit, Bambu users are in a better position to make use of this automatic feature - even more than Prusa users.  I think it would be a great addition to Bambu Slicer.

Attached is a screenshot showing the message in the lower right corner.
[image]

lanewei120, 2023-06-10:
how about paint these logo in the painting by yourself?
[image]
we have provided the height range which can allow you to paint a height range to a special color

it is more flexible than this way

sprior, 2023-06-10:
Yes this is possible several ways, but the PrusaSlicer feature to detect this type of print, suggest it automatically and then perform the color change automatically is super convenient and there are no better users for this than Bambu AMS owners.

lanewei120, 2023-06-10:
I have tried this function on PrusaSlicer
I just mean do it like that way is not feasible as the above picture what I sent

you can not select which color and that will affect all the layers not only this model
for example, there are two object there, it will affect other model's color
[image]

anyway, thank you very much for this suggestion
we can have more discussion

sprior, 2023-06-10:
Multiple models for this is not really an issue - this use case is a layer based color change which is fast because it avoids more than one color printed on any layer.  Multiple models at once isn't really in that use case.

I just tried the Bambu Slicer height based layer tool and it is not nearly as easy to use.  In the test case attached it is tricky to select at the base of the lettering because that's the same height as the top of the flat part so it's difficult to select just the text and not the top layer of the flat.  In my example the intent was to paint just the lettering red and not the top surface.   Prusa's feature makes this effortless.
[image]

sprior, 2023-06-10:
If the select by height feature allowed you to select a vertical surface on the text and then change the color for the range of heights in that surface that would be a good compromise.

lanewei120, 2023-06-10:
> Multiple models for this is not really an issue - this use case is a layer based color change which is fast because it avoids more than one color printed on any layer. Multiple models at once isn't [...]

if the height range is not accurate,
it is still very easy for you to do a bit more
[image]

sprior, 2023-06-10:
I'm back to my original point that the PrusaSlicer feature is automatic and effortless.  Since Bambu Studio is a fork of that code base I would have imagined it would be fairly easy to backport the feature across.

lanewei120, 2023-06-10:
yes, it is easy for us to port the feature, it is not related with the work effort, 
I just want to discuss a more suitable way

thanks very much for the suggestion
we will have more discuss internally

sprior, 2023-06-10:
Keep in mind that you've been selling a lot of printers to people that have never owned a printer before and for whom manually painting colors is a lot for them to grasp.  The Prusa style interaction of "it looks like you just loaded a sign, would you like me to color it for you?" and all they have to do is say "yes", cannot be underestimated as a source of wonder and delight for these users.

dimitarsp, 2024-11-21:
+1 for this feature request. I make a lot of signs and labels. And slicing in Prusa Slicer with the automatic material change is so much faster and more seamless than having to always add height range modifier manually. Especially, when experimenting with different process settings and layer heights, which would result in further tedious adjustments of the layer heights. Painting is out of the question as for single object prints, which would be even more tedious than adding height layers

JohnSmithCYM, 2024-11-26:
In my opinion, this feature was designed to solve the problem of giving text a different color. There are two ways to achieve this:
- Insert text directly in Studio: If you use this method, switching colors is straightforward, especially when using your keyboard.
[image]
- Insert an STL or step file with a text mesh: I have tested this in Prusa 2.8.0, but it seems that the software doesn’t always recognize the text correctly, as shown in the following picture. In my tests, it only provided a "good" result for text located on the top of the model.
[image]
Considering what lane.wei posted, I am unsure if there are other situations we haven’t covered yet.

dimitarsp, 2024-12-05:
I usually add a height range modifier. But that needs to be adjusted each time, especially if using different height profiles. I also have to redo this for each object. Slicing and color painting by layer doesn't work as it still produces unecessary color changes. This is a lot of work in comparison to how seamlessly and easily it works in Prusa Slicer.