bambulab/BambuStudio #1912: Total print hours on printer

bl4ckm0onk3y, 2023-06-08:
**Is your feature request related to a problem? Please describe.**

P1P Printer does not have total hours printed information in printer settings, to ensure correct servicing of the printer (for example each 500 hours of  print time etc) users need to check the total print time. 

**Describe the solution you'd like**
Please add an information in the LCD Panel on the device that contains total hours printed.

sodachen22, 2023-06-13:
Total print hours are shown in X1, there is a plan to support it in P1P.

Hologos, 2023-08-16:
> Total print hours are shown in X1, there is a plan to support it in P1P.

@sodachen22 Is it still planned?

bl4ckm0onk3y, 2024-05-05:
feature was added within the "input shaping firmware update"

closed 2024-05-05 by bl4ckm0onk3y