bambulab/BambuStudio #1913: Remember last filament used - Less Purge

Ruiaxe, 2023-06-09:
**Is your feature request related to a problem? Please describe.**
This is a waist reduction request. If the last used filament, in the previous print is the same of the next print, then the purge need is less.

**Describe the solution you'd like**
A reduction of the purge poop.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Ruiaxe, 2023-06-09:
Opened in the wrong github.

richard-scott, 2023-08-12:
I was just going to post an issue like this. I'm printing from the AMS but the same slot and the same roll of filament each time. Each time I start a print I have to wait for it to purge out he same black filament. such a waste.