bambulab/BambuStudio #1916: Adjusting print speeds doesn't work.

CSWP-Spark, 2023-06-10:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 11.6.2

### Printer

X1C

### How to reproduce

Start a new project. Load a model, pick a preset, and adjust print speed. Slice. Print.

### Actual results

Print speed does not change from the preset speeds.

### Expected results

Print speeds should be adjusted based on user input.

### Project file & Debug log uploads

[BLADE.zip](https://github.com/bambulab/BambuStudio/files/11711451/BLADE.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bambu123, 2023-06-12:
Which parameter did you change?
[image]

SaltWei, 2023-06-15:
The printing speed is too high and already reachedc the max volumetric speed in filament setting.
This is not a bug.

SaltWei, 2023-06-15:
By the way, the cooling logic also make speed lower than settings for small layer.
[image]

closed 2023-06-15 by SaltWei

CSWP-Spark, 2023-06-15:
I slowed the print speeds down. The printer did not slow down. This is a bug.

SaltWei, 2023-06-15:
[image]
Speed is normal in my slicing.

CSWP-Spark, 2023-06-15:
If I load the provided .3MF the following speeds have been altered: Outer wall to 40, Inner wall to 50, and Internal solid infill to 50.If I run with those settings, the machine reverts to the default speeds of 120, 150, and 150, respectively.

SaltWei, 2023-06-15:
> the machine reverts to the default speeds of 120, 150, and 150

The speed in gcode is still 40, 50, 50， right？

SaltWei, 2023-06-15:
I don't understand how machine reverts the speed. Do you means that the real running speed of printer is measured to be 120,150,150?

CSWP-Spark, 2023-06-15:
Yes. I change the speed in the slicer, but the program runs at the default speed from the profile.In order to run this at the speed I wanted, I had to use OrcaSlicer.

SaltWei, 2023-06-15:
Could you share one sliced gcode so I can run it on my printer?
If the speed is 40, 50, 50 in gcode, then it should not be slicer issue.

CSWP-Spark, 2023-06-15:
I’m out of town for a week. I’ll post one when I’m back.