bambulab/BambuStudio #1917: BUG: Behavior of AMS after a print

moebis, 2023-06-10:
### Bambu Studio Version

1.6.2

### Where is the application from?

Bambu Lab github releases

### OS version

macOS

### Printer

Bambu Lab X1C with AMS

### How to reproduce

Wouldn't it make sense for the printer to keep the filament in the toolhead after a print instead of cutting it and purging? I usually end up sending another job with that uses the same filament, it's a waste of material and time. If I happen to change the color on the next print, THEN it could purge and load the new one. Seems more logical to me, unless I'm missing something here.

### Actual results

The printer ejects the filament after every print, even when you're going to send another job using the same filament slot. Waste of time and material.

### Expected results

Don't do the filament eject procedure unless the next job has switched AMS slots.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

shayded-exe, 2023-06-13:
An option on whether or not to unload at the end would be great.

Additionally, the printer should be able to remember what filament it last printed (what's still in the nozzle). That way it can determine whether or not to purge during startup.

mechengineermike, 2023-06-18:
Yours is a feature request rather than a bug, but there is an immediate fix for you.  I have 2 new X1-C's and observed an issue at the end of print jobs where the filament failed to extract from the print head. Modifying the end of job gcode to commend out the extraction solved my problem and would do what you want as well. 
 My problem described here for anyone else readin g this:  https://www.reddit.com/r/BambuLab/comments/zvoind/dreaded_failed_to_pull_back_filament_error/  
 The fix: [image]

Blackboxxxx, 2023-10-15:
This behavior is so dumb. I print tons of items with the same filament.
That you can´t swich unload filament after print off, is a shame.

pichen2021, 2024-01-16:
@moebis 
When AMS is idle after printing, the RFID identification function needs to sent filaments out of AMS, so we unload filament at the end of the print；and it is more convenient for users to replace filaments.
We are evaluating this feature，Thc  X1 firmware is expected to support this feature first.

GuzziRaz, 2024-01-19:
> the RFID identification function needs to sent filaments out of AMS
>
> Thc X1 firmware is expected to support this feature first.

This is simply not true! I have run with the Machine End G-code modified so it never unloads, for half a year without a single problem.

- The printer always knows what filament is (still) loaded - it does so by means of the filament sensors. 
- It does so even after a power cycle. 
- It knows what to do if I start a print with the same color (it just starts printing) or some other color (it automagically unloads the filament and loads the required one).

> so we unload filament at the end of the print；and it is more convenient for users to replace filaments.

A lot of us disagree. The (very) few times I need want to remove an AMS spool that is loaded, or switch to an external spool, I simply hit "unload".

No firmware changes are needed - you just need to add a checkbox in the Print dialog that sets a new placeholder, eg. `unloadAfterPrint`. Then do this:
```diff
--- current	2024-01-19 04:34:36
+++ proposed	2024-01-19 04:37:53
@@ -14,15 +14,17 @@
 M106 P3 S0 ; turn off chamber cooling fan
 
 G1 X100 F12000 ; wipe
-; pull back filament to AMS
-M620 S255
-G1 X20 Y50 F12000
-G1 Y-3
-T255
-G1 X65 F12000
-G1 Y265
-G1 X100 F12000 ; wipe
-M621 S255
+{if unloadAfterPrint}
+    ; pull back filament to AMS
+    M620 S255
+    G1 X20 Y50 F12000
+    G1 Y-3
+    T255
+    G1 X65 F12000
+    G1 Y265
+    G1 X100 F12000 ; wipe
+    M621 S255
+{endif}
 M104 S0 ; turn off hotend
 
 M622.1 S1 ; for prev firware, default turned on
```
Please implement this. Please make that checkbox remember the last used setting. This way all users will be happy.

pichen2021, 2024-01-23:
Hi @GuzziRaz ,
    This feature include two parts:
1. Save time:
  It needs to adjust the end gcode as you mentioned，and this part of the start gcode will detect the filament through the sensor to determine whether to change it. and it affects RFID functions  "Update insertion detection",  "Update on startup" and "Update remaining capacity".
[image]

2. Save filament:
This part of the start gcode will pruge the old filament, not executing this gcode can save about 100mm of filament when using the same filament, but it may cause clog when switching from high to low temperature filament, and may cause color mixing when switching colors.
[image]

GuzziRaz, 2024-01-23:
>    It needs to adjust the end gcode as you mentioned，and this part of the start gcode will detect the filament through the sensor to determine whether to change it. and it affects RFID functions  "Update insertion detection",  "Update on startup" and "Update remaining capacity".

Yes it is a slight downside that with a filament still loaded, when you replace some (other) spool in the AMS, RFID detection will be postponed. I see no problems with that, especially if you default to have this option disabled. Regarding remaining capacity, I saw no problems with it in 5 months (so far) of running with a commented-out unload at the end. Most of the time I only commented out the `M620 S255` and nothing else but nowadays I comment out a little more, just like @mechengineermike [wrote above](https://github.com/bambulab/BambuStudio/issues/1917#issuecomment-1596293617).

>    This part of the start gcode will pruge the old filament, not executing this gcode can save about 100mm of filament when using the same filament, but it may cause clog when switching from high to low temperature filament, and may cause color mixing when switching colors.

So you should execute that code if the filament type was changed, but not if it's the same.

RikshaDriver, 2024-12-24:
Is there a way to check via gcode if a filament change is required?

That condition and then a retraction + purge could then be added to the start gcode, removing the need to purge with end g-code.