bambulab/BambuStudio #1918: Error failed to pull back filament

Benderrodrigu3z, 2023-06-10:
### Bambu Studio Version

Machine issue

### Where is the application from?

Bambu Lab github releases

### OS version

Machine Issue

### Printer

X1C 4 AMS

### How to reproduce

Hit print to D1 line, oh failed to pull back line in B1… there are lines inside the B1-4 tube lines that broke for moisture… I don’t care print to D1, it’s out side the AMS, okay and? You know how much time is required for the filament in the AMS to leave the AMS to reach the HUB, therefor you know it’s not anywhere near the HUB so stop making me dismantle the AMS to be down because it’s line 1 to the machine, just freakin PRINT

### Actual results

Didn’t print

### Expected results

To Print

### Project file & Debug log uploads

Not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sodachen22, 2023-06-12:
Can you please take a video to show the steps of this issue?

Benderrodrigu3z, 2023-06-12:
How? I already fixed the issueSincerely,Shane FSent from my iPhone13

Benderrodrigu3z, 2023-06-12:
I’m telling yall to fix this firmware issue….Sincerely,Shane FSent from my iPhone13

mechengineermike, 2023-06-18:
Your entire issue is incredibly unclear. They might be able to fix it if you describe in clear English exactly what is occurring, what you observe the machine do, all in order.