bambulab/BambuStudio #1920: build on Arch Linux fails with 'error: declaration of ‘tbb::task& tbb::internal::task_prefix::task()’ changes meaning of ‘task’ [-Wchanges-meaning]'

ajacocks, 2023-06-11:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Built from source by yourself or others

### OS version

Arch Linux x86_64

### Additional system information

```
Memory:
  System RAM: available: 7.76 GiB used: 572.6 MiB (7.2%)
  Array-1: capacity: 8 GiB slots: 1 EC: Multi-bit ECC
  Device-1: DIMM 0 type: RAM size: 8 GiB speed: N/A
CPU:
  Info: 8-core model: Common KVM bits: 64 type: MCP cache: L2: 32 MiB
  Speed (MHz): avg: 1992 min/max: N/A cores: 1: 1992 2: 1992 3: 1992 4: 1992
    5: 1992 6: 1992 7: 1992 8: 1992
Graphics:
  Device-1: driver: bochs-drm v: N/A
  Display: server: X.org v: 1.21.1.8 with: Xwayland v: 23.1.2 driver: X:
    loaded: modesetting gpu: bochs-drm tty: 80x24 resolution: 1280x800
  API: OpenGL Message: GL data unavailable in console and glxinfo missing.
Drives:
  Local Storage: total: 100 GiB used: 75.81 GiB (75.8%)
  ID-1: /dev/sda vendor: QEMU model: HARDDISK size: 100 GiB

$ gcc --version
gcc (GCC) 13.1.1 20230429
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

### Printer

Bambu X1 Carbom w/ AMS

### How to reproduce

1. **git clone https://aur.archlinux.org/bambustudio.git**
```
Cloning into 'bambustudio'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 46 (delta 27), reused 34 (delta 18), pack-reused 0
Receiving objects: 100% (46/46), 21.37 KiB | 10.69 MiB/s, done.
Resolving deltas: 100% (27/27), done.
```
2. **cd bambustudio**
3. **makepkg -si**

### Actual results

```
[ 37%] Building CXX object CMakeFiles/tbb_static.dir/src/tbb/private_server.cpp.o
In file included from /home/ajacocks/src/bambustudio/src/BambuStudio/deps/build/dep_TBB-prefix/src/dep_TBB/src/old/task_v2.cpp:29:
/home/ajacocks/src/bambustudio/src/BambuStudio/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:263:20: error: declaration of ‘tbb::task& tbb::internal::task_prefix::task()’ changes meaning of ‘task’ [-Wchanges-meaning]
  263 |         tbb::task& task() {return *reinterpret_cast<tbb::task*>(this+1);}
      |                    ^~~~
/home/ajacocks/src/bambustudio/src/BambuStudio/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:222:9: note: used here to mean ‘class tbb::task’
  222 |         task* next_offloaded;
      |         ^~~~
/home/ajacocks/src/bambustudio/src/BambuStudio/deps/build/dep_TBB-prefix/src/dep_TBB/include/tbb/task.h:33:7: note: declared here
[... 87 lines cut ...]
   33 | class task;
      |       ^~~~
make[5]: *** [CMakeFiles/tbb_static.dir/build.make:328: CMakeFiles/tbb_static.dir/src/tbb/pipeline.cpp.o] Error 1
make[5]: *** [CMakeFiles/tbb_static.dir/build.make:342: CMakeFiles/tbb_static.dir/src/tbb/private_server.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:114: CMakeFiles/tbb_static.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/dep_TBB.dir/build.make:86: dep_TBB-prefix/src/dep_TBB-stamp/dep_TBB-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:345: CMakeFiles/dep_TBB.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

### Expected results

Package builds successfully.

### Project file & Debug log uploads

[build.zip](https://github.com/bambulab/BambuStudio/files/11714073/build.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

c2h5oh, 2023-06-11:
This PR fixes this https://github.com/bambulab/BambuStudio/pull/1882

lanewei120, 2023-06-14:
already merged

closed 2023-06-14 by lanewei120