bambulab/BambuStudio #1921: Win版本深色模式bug

Hotinosin, 2023-06-11:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P

### How to reproduce

1.打开软件
2.把系统主题改成深色模式

### Actual results

打开软件的情况下，切换Windows 11 的深色或者浅色模式hui会成这样

[image]

### Expected results

应该全部界面随着主题改变而改变

### Project file & Debug log uploads

NO File

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-06-14:
你好，
下个版本中将会优化，在win11系统上，将不会再收到系统主题影响，而是由偏好设置中的深色模式开关决定。

walterwongbbl, 2023-08-02:
Hi.
Please use the latest version of Bambus Studio

closed 2023-08-02 by walterwongbbl