bambulab/BambuStudio #1923: File / folder changes of working directory freeze bambulab studio

joseRey, 2023-06-12:
### Bambu Studio Version

Bambu_Studio_win-v01.06.02.04

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 home

### Additional system information

CPU: 10th gen
Memory: 32 GB
Card: NVIDIA laptop

### Printer

X1C with AMS

### How to reproduce

Open a 3mf file from a folder
while the Bambulab studio is open, drop a new STL in the folder
close the Bambulab studio and open again

### Actual results

Stuck at starting, every thing looks white except for 2 gray bars on top.  It looks like the splash screen (also white) and covering the left side of the window, is on top of the second gray bar.

### Expected results

Actual bambulab studio opening screen.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-06-30:
I am unable to replicate the issue. Can you confirm if it is reproducible in your environment?

bambu123, 2023-07-06:
@joseRey Was the window moveable in this case. If the windows is not responsable, please help create the dump status of the program.
The dump is created from the Windows Task Manager, by selecting the hung progress and clicking the 'Create Dump file" context menu.

bambu123, 2023-08-24:
@joseRey We have released a new version, does this problem still exist?

joseRey, 2023-08-24:
Hasn't occurred anymore with the new updates.

bambu123, 2023-08-28:
Close it first, and reopen it if there is a new example.

closed 2023-08-28 by bambu123