bambulab/BambuStudio #1925: KDE - display global scale above 150% bug

adolfotregosa, 2023-06-13:
### Bambu Studio Version

ALL

### Where is the application from?

Bambu Lab github releases

### OS version

Arch

### Printer

Doesn't matter

### How to reproduce

Set Kde display global scale above 150%

### Actual results

(bambu-studio:6903): Gtk-CRITICAL **: 16:08:47.391: gtk_widget_set_size_request: assertion 'width >= -1' failed
Fatal glibc error: malloc.c:2593 (sysmalloc): assertion failed: (old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)
Aborted (core dumped)

### Expected results

Software should open

### Project file & Debug log uploads

I can't

### Checklist of files to include

- [ ] Log file
- [ ] Project file

cmguo, 2023-06-30:
I can't reproduce it. Can you help debug with the gdb tool, and send back the crash stack.

Steps: 
1、gdb Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage
2、type the run command in gdb
3、when it crash and gdb stops, type bt  command in gdb
4、send back all the output

adolfotregosa, 2023-06-30:
> I can't reproduce it. Can you help debug with the gdb tool, and send back the crash stack.
> 
> Steps: 1、gdb Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage 2、type the run command in gdb 3、when it crash and gdb stops, type bt command in gdb 4、send back all the output

Sure:

(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at pthread_kill.c:44
#1  0x00007ffff141a2d3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff13caa08 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff13b3538 in __GI_abort () at abort.c:79
#4  0x00007ffff13b42db in __libc_message
    (fmt=fmt@entry=0x7ffff152fad0 "Fatal glibc error: %s:%s (%s): assertion failed: %s\n")
    at ../sysdeps/posix/libc_fatal.c:150
#5  0x00007ffff13c3227 in __libc_assert_fail
    (assertion=assertion@entry=0x7ffff1531570 "(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)", file=file@entry=0x7ffff152bf14 "malloc.c", line=line@entry=2593, function=function@entry=0x7ffff1531de0 <__PRETTY_FUNCTION__.8> "sysmalloc") at __libc_assert_fail.c:31
#6  0x00007ffff1426ecc in sysmalloc (nb=nb@entry=3088, av=0x7ffff156cba0 <main_arena>) at malloc.c:2593
#7  0x00007ffff1427e93 in _int_malloc (av=av@entry=0x7ffff156cba0 <main_arena>, bytes=bytes@entry=3072)
    at malloc.c:4393
#8  0x00007ffff142872d in __GI___libc_malloc (bytes=3072) at malloc.c:3305
#9  0x00005555588b40ac in wxImage::Create(int, int, bool) ()
#10 0x00005555564e3738 in Slic3r::GUI::BitmapCache::insert_raw_rgba(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, unsigned char const*, bool) ()
#11 0x00005555564e4ccb in Slic3r::GUI::BitmapCache::load_svg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float) ()
#12 0x00005555568de9c1 in create_scaled_bitmap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, wxWindow*, int, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) ()
--Type <RET> for more, q to quit, c to continue without paging--c
_traits<char>, std::allocator<char> > const&, int, bool, bool) ()
#14 0x0000555556828613 in Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
#15 0x0000555556829428 in Slic3r::GUI::TabPrint::build() ()
#16 0x0000555556823228 in Slic3r::GUI::Tab::create_preset_tab() ()
#17 0x00005555566c52a9 in Slic3r::GUI::MainFrame::add_created_tab(Slic3r::GUI::Tab*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#18 0x00005555566c53f8 in Slic3r::GUI::MainFrame::create_preset_tabs() ()
#19 0x00005555566d24bd in Slic3r::GUI::MainFrame::init_tabpanel() ()
#20 0x00005555566d37f8 in Slic3r::GUI::MainFrame::MainFrame() ()
#21 0x00005555566934a8 in Slic3r::GUI::GUI_App::on_init_inner() ()
#22 0x000055555669469a in Slic3r::GUI::GUI_App::OnInit() ()
#23 0x0000555558b99617 in wxEntry(int&, wchar_t**) ()
#24 0x000055555665c14b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#25 0x0000555555e162b7 in Slic3r::CLI::run(int, char**) ()
#26 0x0000555555d71b99 in main ()
(gdb)

bambu123, 2023-07-03:
It seem same as #1868.

bambu123, 2023-07-03:
@adolfotregosa  Are you sure it relate to the X11 scale? Does it crash on 100% scale?

adolfotregosa, 2023-07-03:
> @adolfotregosa Are you sure it relate to the X11 scale? Does it crash on 100% scale?

It works up to 150% globalscale. Above that it crashes.

adolfotregosa, 2023-07-03:
> > > @adolfotregosa Are you sure it relate to the X11 scale? Does it crash on 100% scale?
> > 
> > 
> > It works up to 150% globalscale. Above that it crashes.
> 
> I have made a patch. Can you help test this patch? [0001-FIX-STUDIO-3459-STUDIO-3065-use-FromDIP-instead-of-e.patch.txt](https://github.com/bambulab/BambuStudio/files/11935695/0001-FIX-STUDIO-3459-STUDIO-3065-use-FromDIP-instead-of-e.patch.txt)

sorry I dunno how to compile BambuStudio under arch linux .

bambu123, 2023-07-03:
> sorry I dunno how to compile BambuStudio under arch linux .

OK, please wait the release of new bate version.

If you like to have a try, follow this wiki.
https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.

closed 2023-07-11 by bambu123