bambulab/BambuStudio #1927: Tree supports generated sometimes misses one layer

tangcla, 2023-06-13:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Slice project with tree supports. This only seems to occur sometimes, on certain projects.

### Actual results

When slicing the project, one layer of tree supports is either missing, or shifted to the side . This causes the print to fail in higher up layers.
Below are screenshots to show the same project, sliced differently (different tree style supports). Rotating the project does not seem to help. Changing the support style sometimes helps (ie. hybrid or strong MAY fix the problem) but doesn't always. In this project, it just causes the missing/misaligned layer to be at a different layer height.

 [image]
[image]
[image]

### Expected results

The tree support 'trunks' should be printed with every layer intact.

### Project file & Debug log uploads

[DryPod with Hygrometer v8.zip](https://github.com/bambulab/BambuStudio/files/11739937/DryPod.with.Hygrometer.v8.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Majestic7979, 2023-06-16:
I wonder if this is the same problem I had when I reported this issue: https://github.com/bambulab/BambuStudio/issues/1677

pb4ugoout, 2023-06-18:
I ran into this same issue on the same model. I googled "x1c tree support shifted in middle" and ran across this git issue. 

Really just chiming in to mention that I'm using MacOS 13.5 with Bambu Studio 1.6.2.4 and I have the same artifact in the same location with the same model. I'm using a 2021 M1 Pro 14" Macbook Pro. Seems platform agnostic though.

I wonder if there's something odd about this particular model that is causing it since I haven't had the issue crop up with tree supports on any other model.

tangketan, 2023-06-29:
This bug has been reported before and already fixed. Please wait for our new version which should be very soon.

closed 2023-08-02 by tangketan

KPuzaks, 2024-06-27:
Hello. The issue is present again. Furthermore tree supports are generated through a model. S
[image]
[image]
[image]
ee pictures attached.

Majestic7979, 2024-06-27:
> Hello. The issue is present again. Furthermore tree supports are generated through a model. S ![floating tree support1](https://private-user-images.githubusercontent.com/174014683/343752695-db111642-7473-47e7-b5cd-da55e0afa06f.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk1MTcyMjYsIm5iZiI6MTcxOTUxNjkyNiwicGF0aCI6Ii8xNzQwMTQ2ODMvMzQzNzUyNjk1LWRiMTExNjQyLTc0NzMtNDdlNy1iNWNkLWRhNTVlMGFmYTA2Zi5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyN1QxOTM1MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NjY2Mzc0MzMzN2JkYmZiYTk3NWNmNThmYWYyYmIxMjNiYTU0OGY5ZDllMzg4N2RlYjE5MWIwOTA1YzRmMmIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.AQA8E9FC57BbjsaH_Iy0fzqVDrre3GyVzNnyZeBsxiA) ![floating tree support2](https://private-user-images.githubusercontent.com/174014683/343752707-af316791-3334-4cb7-9e8d-61fe278519ba.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk1MTcyMjYsIm5iZiI6MTcxOTUxNjkyNiwicGF0aCI6Ii8xNzQwMTQ2ODMvMzQzNzUyNzA3LWFmMzE2NzkxLTMzMzQtNGNiNy05ZThkLTYxZmUyNzg1MTliYS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyN1QxOTM1MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OWY4ZGJkNGUwZTljMmYzNjRlOWNiNDFjZGI1MjA5NzUzMzMwZjc4NzM2M2QwMDU4NzU5NDk0MTdkZmJjMWZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.CDUJyWWe71ecTJaerMdzGbazSSGYmeEIL2UMdGFyDEY) ![tree support through model](https://private-user-images.githubusercontent.com/174014683/343752715-3ea247db-cb8e-40d5-8535-ed39eb537396.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk1MTcyMjYsIm5iZiI6MTcxOTUxNjkyNiwicGF0aCI6Ii8xNzQwMTQ2ODMvMzQzNzUyNzE1LTNlYTI0N2RiLWNiOGUtNDBkNS04NTM1LWVkMzllYjUzNzM5Ni5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyN1QxOTM1MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lOTkzOWE1OTY3MmM3MjZjYmQwNzA4ZWZhNTliMjRjMTNiNTkyMjk0ZmZhOTgwNWY1YTczYzBmYmEzYTA4NjZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.YMMtimU9h7R27Q2FnYC7Ixr7mfEVR6RfbQDAYQ4SCXs) ee pictures attached.

I think you must open a new issue. You can reference this. But because you're not OP you cannot reopen directly and because it's closed there is no guarantee that BL will see your report or act on it, please open new issue, reference this one if you want.