bambulab/BambuStudio #1928: Bridging / Tree Support Issues

LegacyOfBob, 2023-06-14:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10

### Additional system information

Processor	Intel(R) Core(TM) i9-10850K CPU @ 3.60GHz   3.60 GHz
Installed RAM	64.0 GB
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab X1C

### How to reproduce

1. Use some model that has a bridge not requiring support (less than 10mm for default)
2. Use any standard print profile
3. Support type "Tree(auto)"

WORKS CORRECTLY IN 1.5.0.60

### Actual results

Bridge direction is at 45 degree angle instead of 0
Bridge will be supported even if less than "Max Bridge Length"
Supports will be missing some layers

WORKS CORRECTLY IN 1.5.0.60

### Expected results

Bridges less than 10mm (or whatever value is in "Max Bridge length" should not be supported
Bridge should be 0 degree angle
Supports should not be missing layers
[image]
[image]
[image]
[image]

### Project file & Debug log uploads

[ForBugReport.zip](https://github.com/bambulab/BambuStudio/files/11740349/ForBugReport.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-06-29:
Thanks for reporting the issue. The internal version that we are working on is fine with supports. But indeed the bridge angle is not optimal. This may be related to a new feature that we recently added: auto bridge angle. We'll investigate.
[image]

QingZhangBambu, 2023-07-18:
Considering to avoids the put the bridge path in the overhanging wall, that direction is best.

QingZhangBambu, 2023-07-18:
In 0 degrees, the point part may fail
file:///C:/Users/Administrator/AppData/Roaming/LarkShell/OptimizeImage/bf4e3a38-e8b5-4721-96c7-fa026aff6348.jpeg

QingZhangBambu, 2023-07-18:
In 0 degrees, the point part may fail
[image]