bambulab/BambuStudio #1929: [Bug Fan] fan 100% ASA

MarcoPiazza83, 2023-06-14:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

MacOs Ventura

### Printer

X1C

### How to reproduce

ASA
Tab set filament
Max fan 40%
Slicing

### Actual results

By setting ASA, in the filament tab the max speed is 40%.
In the slicing I see some layers with 100% fan.
I will insert photo and video slicing.

[image]

https://github.com/bambulab/BambuStudio/assets/4783556/52ff0030-2a3d-4d3d-a4dd-53b96cd26a57

### Expected results

the maximum fan speed must be equal to the maximum speed set in the filament tab

### Project file & Debug log uploads

[piastrella v4.3mf.zip](https://github.com/bambulab/BambuStudio/files/11743896/piastrella.v4.3mf.zip)

Project File

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-06-21:
Can not reproduce this issue with above 3mf file
[image]

MarcoPiazza83, 2023-06-21:
> Can not reproduce this issue with above 3mf file
> [image]

I can only reproduce your screenshot by turning off the "keep fan always on" flag.
Is the filament profile also exported in the 3mf file?

SaltWei, 2023-06-21:
> Is the filament profile also exported in the 3mf file?

[image]

SaltWei, 2023-06-21:
Also change the option
[image]

MarcoPiazza83, 2023-07-02:
Win or Mac?
I checked again.
In mac it keeps giving me that problem

SaltWei, 2023-07-06:
> Win or Mac?

I am using windows

SaltWei, 2023-07-06:
I will find a mac and try again

MarcoPiazza83, 2023-07-10:
> I will find a mac and try again

confirm, the issue only in Mac Os.
screenshot mac vs win

[image]

[image]

BambooBugFixer, 2023-07-11:
I used your project file and tried it. It works fine on my Mac. 
[image]
What's the version of your Mac? Mine is like this. 
[image]

MarcoPiazza83, 2023-07-11:
> I used your project file and tried it. It works fine on my Mac.
> What's the version of your Mac? Mine is Ventura 13.2.1.

[image]

[image]

BambooBugFixer, 2023-07-12:
I used a Mac with  Ventura 13.4.1 and a bamboo studio with the version of 1.7.0.53. But I still cannot reproduce the same issue. 
[image]
Has any other use with the same environment encountered the same problem?

closed 2024-08-09 by XunZhangBambu