bambulab/BambuStudio #1932: Bambu Stuido Crashes on set of files from OpenScad

timmeade, 2023-06-14:
### Bambu Studio Version

01.06.02.04 

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.4 (22F66)

### Additional system information

Macbook pro  ARM-64 (Native)
[bambuStudioCrash.txt](https://github.com/bambulab/BambuStudio/files/11747835/bambuStudioCrash.txt)

### Printer

P1P

### How to reproduce

I have created two different files of a particular size but different depths.  Both load in Bambu Studio but the moment i slice either it crashes.

I have rebooted and no changes.

Files are proprietary  I cannot post.   I have done hundreds of similar files in past.   But today these two cause crashing.

Hopefully attached crash file will help

### Actual results

Studio crashes moment slice button depressed.

### Expected results

Files should slice.

### Project file & Debug log uploads

[debug_Mon_Jun_12_10_53_34_82672.log.0.zip](https://github.com/bambulab/BambuStudio/files/11747854/debug_Mon_Jun_12_10_53_34_82672.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-06-15:
0   BambuStudio                   	       0x10320cef8 Slic3r::Voronoi::annotate_inside_outside(Slic3r::Geometry::VoronoiDiagram&, std::__1::vector<Slic3r::Line, std::__1::allocator<Slic3r::Line>> const&) + 1128
1   BambuStudio                   	       0x1031efc30 Slic3r::Geometry::MedialAxis::build(std::__1::vector<Slic3r::ThickPolyline, std::__1::allocator<Slic3r::ThickPolyline>>*) + 80
2   BambuStudio                   	       0x103038b98 Slic3r::ExPolygon::medial_axis(double, double, std::__1::vector<Slic3r::ThickPolyline, std::__1::allocator<Slic3r::ThickPolyline>>*) const + 80
3   BambuStudio                   	       0x1032f1668 Slic3r::PerimeterGenerator::process_classic() + 9988
4   BambuStudio                   	       0x10322a628 Slic3r::LayerRegion::make_perimeters(Slic3r::SurfaceCollection const&, Slic3r::SurfaceCollection*, std::__1::vector<Slic3r::ExPolygon, std::__1::allocator<Slic3r::ExPolygon>>*) + 624

SaltWei, 2023-06-20:
Could you please share one crashed stl model for us to debugging?
We have found some abnormal code, and try to fix it. We hope we can get one 100% crashed case to check whether the fix work or not.