bambulab/BambuStudio #1933: A fatal error occurred: "Access violation" when slicing lithophanes if you have a raft

Cetacea37, 2023-06-14:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home

### Additional system information

Processor	Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz
Installed RAM	32.0 GB (31.8 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab X1C

### How to reproduce

1) Create an 80mm x 80mm x 2.8mm lithophane at itslitho.com
2) Load the STL into Bambu studio
3) Add a raft
4) Slice file

### Actual results

Once the slice reaches 35% it pops up an error that says "Fatal error occurred: "Access violation".  Please save your project and restart the application."

If you remove the raft this doesn't happen.  But once it does happen you have to restart Bambu Studio as all slices will have this error.  

Also, this doesn't happen in v. 1.6.1.  

### Expected results

File slices and is ready to be sent to the printer.

### Project file & Debug log uploads

[debug_Wed_Jun_14_07_06_50_6440.log.zip](https://github.com/bambulab/BambuStudio/files/11750011/debug_Wed_Jun_14_07_06_50_6440.log.zip)
[Lithos_Ginger.zip](https://github.com/bambulab/BambuStudio/files/11750013/Lithos_Ginger.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MadMax389, 2023-06-19:
Kind of strange. If I import just the stl and slice it with a .2mm profile with 2 layers of raft, it works fine
[image]

Cetacea37, 2023-06-19:
> Kind of strange. If I import just the stl and slice it with a .2mm profile with 2 layers of raft, it works fine [image]

Not sure what to tell you there.  I had my business partner try at his house on his PC and he had the same problem.  He's the one who realized it was the raft that was crashing it.  I also tried it in the most current version of Orcaslicer and it also had the same problem.  But it works fine on the previous version.  I did like 120 of these in April no problem.

MadMax389, 2023-06-20:
I did try your .3mf and it crashes as you stated. Must be some combination of settings BS doesn't like. Have you tried slicing it with the standard BBL .2mm profile?

Cetacea37, 2023-06-20:
> I did try your .3mf and it crashes as you stated. Must be some combination of settings BS doesn't like. Have you tried slicing it with the standard BBL .2mm profile?

If you just use the standard BBL .2mm profile it slices fine.  If all you change is adding a raft, it crashes.  At least it does for me.

tangketan, 2023-06-28:
This bug has been fixed. New version will be released soon.
For now you can enable support but change to normal(manual) as a workaround.

tangketan, 2023-08-10:
The fix is included in Studio V1.7. Please have a try.

closed 2023-08-10 by tangketan