bambulab/BambuStudio #1936: Own filament manufacturer name

DokuKaefer, 2023-06-16:
Please add a way to customize the filament manufacturer name yourself.
This would greatly increase the clarity in Bambu Studio with different filament manufacturers.
Currently I often have to look up which manufacturer is in which AMS slot.

DokuKaefer, 2023-08-16:
@bambu123 or @sodachen22 
is there already a time frame when we can expect this function?

lanewei120, 2023-08-17:
@DokuKaefer 

Hello, we are adding a way to let user to add third-party filaments in bambustudio
currently it is under design and hope will come out in not a long time

and to help us better understand all the use cases,
could you give more details on your use case?

thanks

DokuKaefer, 2023-08-17:
@lanewei120 

Hello, thank you for your feedback!

I have three Bambu Lab X1C printers here with a total of 12 AMS systems.
Since I always buy filaments from many different manufacturers and then fill them up depending on the empty AMS slot. 
The printers are also located in a different room than my office, so I always have to keep a list of which manufacturer is currently loaded in which slot.

If I could select the appropriate manufacturer directly on the touchscreen when inserting the filament, I would then see this in Bambu Studio and can select the appropriate profile for the filament.

Best Regards!

DokuKaefer, 2023-11-09:
@lanewei120 yeah thank you! 👍

DokuKaefer, 2023-11-09:
@lanewei120 i think there is a bug in the "Preset selection"
I can only select "Generic PLA" but not my existing ones.

[image]

User presets
[image]

lanewei120, 2023-11-10:
[image]
[image]

I can see other filaments here
have you selected these filaments to visible before?
[image]
only the visible filaments can be shown in the drop list

if you can find them in the drop list in the main window
then it should be a bug
and please help to upload the log to us for a debug

thanks~

DokuKaefer, 2023-11-10:
Yes, the filaments are selected.

[image]

LOG File
[log.zip](https://github.com/bambulab/BambuStudio/files/13315865/log.zip)

murongguao, 2023-11-11:
Hello, I'm glad you can use our new features.
There is indeed a problem with the issue that you cannot select the user preset you created when using the Create Filament function.
Here is a solution for you. First, use the system preset to create a new custom preset,
[image]
Then go edit filament page，
[image]
click the "+ Add preset" button.
[image]
Select the printer where you want to add a filler. Then select the filling preset you previously created. You may encounter situations where the preset already exists, click "Yes".
[image]
Finally, the custom Filament preset under this printer will be replaced by your user's preset configuration.

I'm sorry for the inconvenience this issue has caused you. We will fix it as soon as possible.
Salute.

DanBao-Bambu, 2023-11-11:
Hello, I'm glad you can use our new features.
There is indeed a problem with the issue that you cannot select the user preset you created when using the Create Filament function.
Here is a solution for you. First, use the system preset to create a new custom preset,
[image]
Then go edit filament page，
[image]
click the "+ Add preset" button.
[image]
Select the printer where you want to add a filler. Then select the filling preset you previously created. You may encounter situations where the preset already exists, click "Yes".
[image]
Finally, the custom Filament preset under this printer will be replaced by your user's preset configuration.

I'm sorry for the inconvenience this issue has caused you. We will fix it as soon as possible.
Salute.

DokuKaefer, 2023-11-11:
@DanBao-Bambu thank you for your help.

But this workaround is not possible for me either.
In the window "Add preset for new printers" i can not open the dropdown menu.
Please see here:

[image]

DanBao-Bambu, 2023-11-13:
I'm sorry to see your results. There are no issues with your operation, and Studio will fix this issue in the next version. Here are the solutions:
1. Ensure that there are at least two printer models in your selection, and select the other one.
2. Enter the edit filtration dialog and click "+Add Preset"
[image]

I'm sorry for the inconvenience caused to you. I hope it can solve your problem.

DokuKaefer, 2023-11-13:
@DanBao-Bambu perfect! That workaround works for me. Thanks!

DanBao-Bambu, 2023-11-29:
Hello, I will reply to you one by one.
about https://forum.bambulab.com/t/bug-v1-08-62-beta-2-custom-filament-preset-fix-issue/37293 :
Firstly, I know you want to use a better and more suitable name to name your fillment, but the name you create has a meaning. When you use this consumable on the printer's AMS, it will be parsed based on its name, so if you have better preset parameters and want to use it on AMS, please save it with its original name.
Secondly, I think I understand why it appears to have been deleted on the UI, but it still exists. Is your operating environment like this: First, the preset on the main page is selecting the preset you want to delete. When you click delete in Edit Filament dialog, you actually click no, and then it disappears from the interface, but it does exist. This is a bug. I will make repairs.

DanBao-Bambu, 2023-11-29:
about https://forum.bambulab.com/t/beta-studio-v1-8-beta-2-question-on-custom-filament-method/37304/2 :
Regarding Issue2: This is indeed a bug, thank you for raising it. The fields here: "Vendor", "Type", and "Serial" should be consistent with what you filled in when creating, and their combination represents a Filament.
Inherited presets cannot be deleted after having child presets, because the parent preset has basic properties from child presets. Child presets without a parent preset cannot be used, and only custom parent presets can be used by AMS.|
Regarding the vendor type serial for custom Filaments, this is to consider the consistency design of AMS. The name of the printer with a longer preset name is also designed to match the corresponding printer and distinguish it from the system preset.

DanBao-Bambu, 2023-11-29:
@OzDruiD About https://forum.bambulab.com/t/custom-filament-issues/36350/29 :
There is too much content here. Can you tell me where you still have any questions? I can reply to you on GitHub. Thank you.

OzDruiD, 2023-11-29:
> Inherited presets cannot be deleted after having child presets, because the parent preset has basic properties from child presets. Child presets without a parent preset cannot be used, and only [...]

Ok , I understand that, I dont love it, as we are forced to have both a Parent preset - (without all the necessary properties, or even overwritten with some, such as basic nozzle temps),    IF I create a custom preset (based off parent ) - and we name it appropriately to actually mean something, we are still stuck with that Parent in our drop down \daily use list .., which eventually list becomes MASSIVE.    

At some point, we need the ability to "sort\hide" parents or other presents from regular Slicer drop downs (to clean up the massive list) - without actually deleteing them (or their assocated presets or flow calibrations)  in case we want to use in future . Hiding child \ parents should also be independat of each other, but still able to use any features from each other ( eg its just hidden)    eg  Hide the parent .. eg  "Sunlu PLA Anti Stringing" as another example   but my Sunlu PLA AntiString - high flow -0.4 BBL X1C  is still shown.   Its vcendor, material, printer AND important descriptor (and would still inherit the parent presets)

OR - at least a customisable "favouriting" method - and we favourite our custom modfiied presents, which moves to the top of the list-  and the junk parent unused presets are a the bottom.

Sounnds like a feature request now, but its a part of the issue with the current implementation of Custom filaments.

OzDruiD, 2023-11-29:
> Can you tell me where you still have any questions? I can reply to you on GitHub. Thank you

Ok sorry- yes. Its all around the implemenntation of this Custom Filaments deature, so a few questions based on your previous replies in sections below to seperate, shown with your quoted replies @DanBao-Bambu    -Thank you for your patience.

------------
> so if you have better preset parameters and want to use it on AMS, please save it with its original name.
and
> and only custom parent presets can be used by AMS.|

a) Do you mean save over the inital custom filament name then ?  
eg modify the nnozzle temps, colour etc- and then save without changing the name?  a  eg the fulll long name "SunLu PLA + @ Bambu Labs X1 Carbon 0..4 Nozzle"    (What I call the "parent")

b) Are you saying this master \ Parent will be the the only one available to AMS?  
- If so, and I want the AMS \ printer to see multiple Sunlu PLA, in different colours, different flows - I need to save a "parent" for each filment, and modify those "main - long names?" 
 - If so, the what are child presets usefull then if cant be used by AMS?  The flow  \ k value drop down section only?
 
 Apologies if I am mis understanding how we are meant to use Custom, Parent\Child presets vs Filaments etc,  Please explain your expected view of how this should work.

---------------------- 

> When you click delete in Edit Filament dialog, you actually click no, and then it disappears from the interface, but it does exist. This is a bug. I will make repairs.

Thank you, yes I think you are on the right track, but I am not clicking NO,  I actually click "Yes\OK\agree" whatever it is to confirm the deletion - it trieds to do the delete, it hides and deletes it, but its doesnt really..  Will let you review and advise of fix\repairs.
-------------

OzDruiD, 2023-11-30:
@DanBao-Bambu  I apprecaite the effortgs, I fully understand how to name and customise names - its the duplication and labourous LIST that is frustrating.  I want to hide \ remove the unused presets\filaments ., but not delete from my "database" for future use.
2, or even 3 different "filaments" is understandable for the same product if you have custom settings - fine

[image]
This is just a simple example .. but the bah blah Blah @  Parents ones...  (RED BOX) - I would never use as they dotn have all the settings.
[image]

Are you sayinng .. just sasve over the (RED) parent ones, and not create additional custom ones to minimise my list?
1) I can have a shorter list - without duplicates if I dont change the save name and create new custom ones
2) If I want the shorter better usable names, I have to deal with the duplicates \ longer list  because the parents will be shown?

Is the above essentially accurate for now until features\fixes in future?

DanBao-Bambu, 2023-11-30:
@OzDruiD 
> a) Do you mean save over the inital custom filament name then ?
> eg modify the nnozzle temps, colour etc- and then save without changing the name? a eg the fulll long name "SunLu PLA + @ Bambu Labs X1 Carbon 0..4 Nozzle" (What I call the "parent")
Yes, your understanding is correct.

> b) Are you saying this master \ Parent will be the the only one available to AMS?
>
> * If so, and I want the AMS \ printer to see multiple Sunlu PLA, in different colours, different flows - I need to [...]

Indeed, only the parent preset can be used for AMS, as its name is the most standard and in line with the design. We just want to display the correct Filament like "Bambu PLA Basic" on your AMS when using it. In fact, you can still create your favorite sub presets on Bambu Studio, which does not affect the process of slicing and printing. Your sub presets can still be used for the corresponding Filament slots on AMS.

Majestic7979, 2023-11-30:
Bambu Lab, all of this would be resolved if only you programmed Bambu Handy to use NFC to program those cheap round tags. You can make a wizard on the app to "add filament", the wizard would ask the brand, the color name, allow user to choose the actual color from a picker, the material, the parameters (temperature etc), and what have you, then bring the phone close to the tag, and record it. Then, the user sticks it on the side of the spool and AMS reads that information just like it does for your Bambu Lab filament spools with RFID.

But instead of being customer-friendly you lock down the appliance that I paid over £300 for, and for what? Your own filament is always out of stock, takes ages to deliver, and from the sample that I used of the orange one it is awful in quality. I never had filament break into pieces but yours did. I have filament from a company called Technology Outlet in the UK, they use a EU supplier for their OEM filament and I have a dozen spools all perfect. Stored in the exact same conditions. But instead of allowing your customers to make use of the hardware they paid for by letting them use cheap NFC tags from ebay, you lock it down to yours.

Do you even ship filament all around the globe? You have dealers in Brazil selling the X1C for R$16,500, that's a massive cost to a Brazilian when their minimum wage is something like R$1,000 per month. And you do not ship your filaments to that region, so they buy a crazy expensive printer with AMS that half the functionality is not even usable. This is insane. Stop locking it down and let people use MiFare tags to program their own spools so you can stop the madness that is entering this info on the slicer for every single roll and then have to manually select it for every print.

It's dumb and you're making it so for no reason. Are you in the printer business or the filament business? Pick your battle. You are not Hewlett Packard selling inkjets for $20 to charge $80 on the cartridges! The AMS is mostly plastic and it costs a whopping £300+ per unit YOU MUST STOP CRIPPLING IT INTENTIONALLY!

I want to put MiFare tags on my own filament spools of whatever brand I choose and have the AMS know what it is. Practically all phones now have NFC chips inside for payments so everyone has a way to program these tags. Even if the AMS cannot "write" to the tags (which we all know is not true, it can write) then Bambu Handy could serve as the tag programming device on the user's mobile phone. Bonus: it can send the info to your cloud and have it ready to pick from the printer. Then when the user puts that spool in the AMS: BOOM it's recognized like magic, just like it happens with your proprietary low-quality filamnent... As for where do the tags should go on the spool? We are smart enough to design templates to help fixing in the right place. You can provide dimensions from the center of the spool for alignment.

If there is a will, there is a way. Use what you have to your advantage and stop alienating your customers. You are starting to make printing too complex. Judging by your corny video on your site of a boy and his father printing model train tracks like the thing is an oven baking a pizza, you are really starting to drift off from that ideal/mission. The slicer is now becoming a huge mess. If your business model is about expanding 3D printing to the masses you should use what you already have to simplify the process, just look at the screenshots above, it's a mess!

OzDruiD, 2023-11-30:
> #1 and #2 are basically correct. And "Vendor Type Serial" is necessary for the use of AMS.

Yes, I see - what I wanted is not going to happen - shorter names without the @ .. or hide the parent presets from the Slicer
- I mean you do it onn the AMS - its just the parent, with the presets hidden.. so I guess we have to operate like that in slicer.. and somehow.. deal with it....

Speaking of AMS

> We just want to display the correct Filament like "Bambu PLA Basic" on your AMS when using it. In fact, you can still create your favorite sub presets on Bambu Studio, which does not affect the [...]

Hmm, dont think this is working well in BETA
My presets \ Parent filaments - arent working properly but I think maybe you know of this bug already?.
1. Filaments arent linking their flow conntrol calibrations to their parent Filament for dispaly in AMS, tthis drop down should have a bunch of presets under Dynamic control - but nothing works
[image]

2. Separetly - some Filament sub-presets with their flow controls are being linked to "Generic PLA" 
-> Notice the Dynamic Pressure control drop down
[image]
2a. Note the WOOD control value isnt a preset\filament -at this stage, its just a calibration for Sublu - but its in AMS printer screen as Generic PLA???
[image]
2b. For comparison. Slicer Flow Dynamic results....## Wood Calibration linked to Generic PLA?
[image]
2c. Slicer AMS selection also matches AMS screen on printer
[image]

3. Vendor\Serial - in this last exmaple -we have a upper case SUNLU -with use of Material annd Serial - defining Parent filaments, shown on the AMS as per your explaination - Great!   I think this matches you design!!
[image]
3a.   These Glow, etc etc are Serials - create Parent , AMS available Filaments  fair enough. Great!
3b.  Dynamic Calibrations are where?   gone .. somehow lost .... not in slicer flow calibrations either.
[image]

3c.  Technically, if added as a Parent filament, like Glow
eg SunLu - Wood PLA  ( not PLA+)  - it should appear under the SunLu Vendor - as a seperate matrial + serial Typer.
- it doesnt - only PLA+   is shown (and as above - Dynamic flows are missing\lost even for the +) 

I propose I will pause here, wait for BETA 3 release, delete all my finlaments, calibrations and make a fresh start after your next round of fixes, following your  Parent  Filamentr -    Vendor - Material - Serial method... and report back \raise a new ticket at that time .. becuase....  this is just looping around and not making sense, or showing up properly, across Slicer and Printer (AMS) screens, with all relevent calibration data linked to the correct Filament types, even when I try and follow that Vendor - Material - Serial method.

Thanks for efforts.  - if you do wish me to test things, happy to so - but yeah ... feels like until next Beta with filament fixes, this might be a waste of time.

DanBao-Bambu, 2023-12-01:
Thank you for pointing out our issue.
> Dynamic Calibrations are where? gone .. somehow lost .... not in slicer flow calibrations either.
> [image]
Regarding this issue, please confirm if you have deleted the Filament from the Studio after calibration. At this point, the printer is unaware that the material has been deleted. This issue is being fixed, but I don't know when it will be fixed and released.

> Note the WOOD control value isnt a preset\filament -at this stage, its just a calibration for Sublu - but its in AMS printer screen as Generic PLA???
It's not like that, our expectation is to place the calibration preset on its corresponding Filament. When saving the calibration preset, we will send the relevant information to the printer. I don't know what happened on the printer that caused it to be placed on the Generic PLA. Is it possible that the printer was not aware of the newly created custom Filament because you did not synchronize the printer after creating the custom Filament? Can you provide the printer's log for developers to see what exactly happened on the printer?

OzDruiD, 2023-12-01:
Both these issu

> I don't know what happened on the printer that caused it to be placed on the Generic PLA. I

This has already been logged by various people and logs submitted through the bambu foruims I see so as I already deleted all my filaments, resycned lean and refreshed - I can "Try" and reproduice if you still need it .. but thread here https://forum.bambulab.com/t/custom-filament-issues/36350/39

OzDruiD, 2023-12-02:
2ND TEST - Create a 'nested'  new cusstom filament, eg based of an EXISTING CUSTOM FILAMENT (SUNLU PLA +)
- Vendor - SUNLU
- MATERIAL - PLA
- Changed serial to + GREEN  
- Changed name to PARENT 2
[image]

Do the calibration (parent 2) annd verify how that calibration values are linked to the Filament\Presets.
- got an error it exceds 40 - so had tro rename a bit (fair enough - printer LCD space I assume)
[image]
- renamed shorter 
[image]
[image]

You can see here - the SAVED flow calibration result - is correctly saved to the SUB-PRESET.
[image]

SUNPLU PLA + GREEN PARENT 2  -->   SAVED TO  -->  SUNLU PLA  + GREEN (and confirmed in the edit box)
[image]

This is what it should do for ALL cusstom filaments\sub-presetrs - not save ssome to Generic PLA.

OzDruiD, 2023-12-03:
@DanBao-Bambu 
> Sorry, there was something I couldn't reply to yesterday. Today, I will keep an eye on your issue and handle it seriously.
No problem at all, please look after yourself, reset , family etc :P 

> I need to know the printer version and your Studio version.
Certaily, I am on the BETA path for both my X1C Carbon and Studio software.

Studio BETA - 1.8.0.62 / Plugin is 01.08.00.03
[image]

Printer and AMS (I removed serial)
Printer - 01.06.06.03 (Beta) /  AMS - 00.00.06.40
[image]

DanBao-Bambu, 2023-12-03:
Now my Studio and printer versions are the same as yours.
[image]
[image]
I created "SUNLU PLA+" based on "Generic PLA" and obtained a parent preset "SUNLU PLA+@ Bambu Lab X1 Carbon 0.4 noise". Afterwards, I created a sub preset named "SUNLU PLA+GREEN @ BBL X1C 0.4" based on this parent preset.
[image]
Afterwards, I set my AMS slot 3 to Generic PLA.
[image]
Then select the previous sub preset "SUNLU PLA+GREEN @ BBL X1C 0.4" for calibration.
After calibration, save it directly and the information you see on the calibration results page is correct.
[image]
[image]
Unfortunately, your problem was not reproduced.
Can I take a look at the editing page of the Filment you created, as shown below. Confirm if the sub preset you created is under the expected Filament.
[image]

OzDruiD, 2023-12-03:
> Afterwards, I created a sub preset named "SUNLU PLA+GREEN @ BBL X1C 0.4" based on this parent preset. [...]

Did you do that inn the "Add preset" buttonn in that screen?

I did Not do that.  To create my preset - I went into 
1, PREPARE screen
2. Selected hte Parent Filament created innitially - SUNLU PLA + @Bambu Lab X1 Carbon 0.4 Nozzle
3. Modified the settings, colour, temps etc,  -
[image]
 then "saved that" with the other name - Green etc  - this should  still make a sub custom preset based off the intial parent preset.
4. Saved as follows - 
[image]

Filament preset screen as requested
[image]

> Afterwards, I set my AMS slot 3 to Generic PLA. [...]

Huh? Why?  Should you not be sselecting the Mater preset you already created to calibrate (not generic?) eg SUNLU PLA + @xxxx etc

DanBao-Bambu, 2023-12-03:
Yes, I created the sub preset according to your method. I modified the colors in the Filament settings dialog and saved it under a different name。
[image]
I set slot 3 of AMS to "Generic PLA" because I want to confirm if this is the reason why your dynamic pressure control parameters were placed on this Filament.
I am now setting slot 3 to SUNLU PLA+and performing calibration. I am waiting for the calibration result.

DanBao-Bambu, 2023-12-03:
@OzDruiD My custom Filament "SUNLU PLA+" was indeed created based on "Generic PLA". But in reality, once you create it successfully, this preset will have nothing to do with "Generic PLA". So I'm confused. Can you send me the parent preset you created so that I can check its content.
This file is located in： %appdata%\BambuStudioBeta\user\user_id\filament\base.
user_id  needs to be replaced with your own.

OzDruiD, 2023-12-03:
> @OzDruiD My custom Filament "SUNLU PLA+" was indeed created based on "Generic PLA". But in reality, once you create it successfully, this preset will have nothing to do with "Generic PLA". So I'm [...]

Sure, I also repeated the process, via greatring a 2nd one - poly Terra etc here
[image]

So can send you.

DanBao-Bambu, 2023-12-03:
@OzDruiD Okay, I will wait for your result. If there are any issues, please provide me with the preset you have. Alternatively, package the entire "user_id/username" directory and send it to me.

OzDruiD, 2023-12-03:
@DanBao-Bambu  Soundns reaally dumb. but how do  i sennd \submit you files?

DanBao-Bambu, 2023-12-03:
This button should be able to send attachments to me.
[image]

OzDruiD, 2023-12-03:
Ok senndinng now - in silcer and printer, those presets are seen under Generic too btw
[image]
[DruiD_Custom_Filamennts_user_Folder.zip](https://github.com/bambulab/BambuStudio/files/13538347/DruiD_Custom_Filamennts_user_Folder.zip)

OzDruiD, 2023-12-03:
PolyTerra calibration finnished - Not Generic PLA (Great!!)  - Wierd??
[image]

Still nothing available in this drop down thou.. for PA Profile or Factor K? 
[image]

OzDruiD, 2023-12-03:
So can you export the printer logs to an SD card and send them to me?
 - Sure but is there a better way than open forums ( or are my logs and personal stuff all protected?)

DanBao-Bambu, 2023-12-04:
@OzDruiD Yes, please give me the "\log" directory

OzDruiD, 2023-12-05:
Howdy all, just to update this offically,
1.  A HUGE thank you to @DanBao-Bambu  - who, has been simply amazing in working with me, explaining, documenting, troubleshooting and outlining steps.

2.  In summary a combination of fixes in BETA 3   -and work flow practises, will fullfil my requirements and will detail to close out here.  BETA 3 - https://github.com/bambulab/BambuStudio/releases/tag/v01.08.01.57

Relevant fixes to my issues from Beta #3 - but all fixes in #3 are great!
- Delete the preset in the editing process, and click No, the preset will disappear in the UI, and it will still be available when reopened.
- Fix the issue that the name of the material in the error prompt is empty when the custom material does not match the printing board.
-  The filament serial may not be correct in the editing filament dialog

_**STANDARD EXPECTED USE OF NEW CUSTOM FILAMENTS**_
A) Correctly Display Preset Calibrations for FILAMENTS in device screen    - bug fixes above
B) Have Custom Filaments, with multiple Presets for varaitions, with the Custom Filament allowing various Calibrations, PA, K vvalues - saved to the Parent Customer Filament for use with the AMS and Slicer - working - as per design

**_MY UNIQUE CASE- RESTRICT END-USER ISSUE BY CUSTOMISING 'SINGLE' FILAMENT VERSIONS_**
*Through some worklow and understanding of process, a special usecase I need is solved.

A) Custom Filament NAMES require to be full  details-  Vendor, Material  - PLUS  Colour, temp range -> Written in **_words_**, for "accesibility and inclusion" reasons like colour blind, young children, or disabled students etc, to make it much easier and 'fool proof' for them.

Example of how its requried to show extra details in the filament screen - note the yellow text vs colour icons as example.
[image]

D) Custom FILAMENT with those specifics required as discussed n C)  will be created by using the Custom Filament Wizard, with unique Vendor, Material type - and serial - TO ENSURE IT GENERATES A _UNIQUE FILAMENT ID_ 

E) PA\Flow Calibrations will be done carefully, to ensure its correclty selecting that unique Custom UNIQUE FILAMENT from D)  

**_and thus_** 

F) this workflow will LIMIT the PA profile to a SINGLE entry as its tied to a specific custom FILAMENT (due to the ID linking)

** This is diferent than normal method of Custfom Filament creation and calibration linking**
Normally you  would have a single master custom FILAMENT, for example
Vendor: SUNLU
Material: PLA
Serial: +

Them you may create various presets 
- SUNLU PLA + Green
- SUNLU PLA + Red
 etc for ease of reading in filament screen with their custom attributes like temp, color, price etc etc configured in that _PRESET_

 ** HOWEVER** 
As per design, Calibrations in this case (PA Profiles) would be assigned to the PARENT - SUNLU PLA + _FILAMENT_  - and this appear in the device drop down as such, grouped via that Parent Filament name
 
[image]

 If you need to _LIMIT IT even further_, eg to remove user error, or  to make it simple\easier,  (again, accessbility, heath issue, young kids)  - this will NOT WORK as multiple calibrations, eg Green, Red will sstill be visable.
 
 The the workaround above, C-E achieves this and thus will limit inexperienced users to 
a) Selecting the properly named fillament - eg  SUNLU PLA + YELLOW
b) only having a single choice in the PA Profile drop down for calibration etc.

Note- there is greater admin overhead here, managing and updating lots more Custom filaments, redoing calibrationss etc, but for my application , young kids, medical limitations etc, its the most appropriate.

**_This TLDR, is but a SUMMARY of much larger work effort by @DanBao-Bambu and myself, supporting my questions to land on a process and bug fixes, that will help me support the people I help that present with challenges and help bring them into the world of 3D Printing and Bambu, so really appreciate this work effort and fixes ... thank you!_**

OzDruiD, 2023-12-07:
Hmm ok .  I think I have to fully revert to secondary method @DanBao-Bambu and create a "Master" custom filament for each and every filament that I want to be "named" properly and flow completly through all the way through to AMS  (and back to slicer).

I missed , in testing, just using the normal design way, that once I sync with the AMS - it restores (in the slicer view screen - the main Parent filament on pull back from the AMS and changes out any defined "presets" from that parent, it makes sense -as the AMS only holds that parent)

My people setup to use the custom presents with the named colour, temp values (remember, they cant SEE the colour, low vision, colour bliond etc, young kids- so they have to read)

So AMS #3 - Bright Green,   AMS #4 - Red 
[image]

As soon as you Sync\resync - it reverts back to that Parent Custom Filament.
[image]

Filament master and presets for completeness sake in this thread so it makes sennse to others reading.
[image]

OzDruiD, 2023-12-08:
K tried the sync thing  - loaded up in the device ams screen some PolyLite PLA - with differnet colours and no PA Profile
[image]

then went to slicer and did a full Resync-  polylite comes across -great
[image]

then changed a filament to the Sunlu+ 
[image]
Did the SYNC
[image]

IF however, before they do the Sync they change the Preset - to the proper named one with other settings etc
and do a SYNC - that doesnt change it  (as you explained)
[image]

So, its still 2 changes.. change physical in AMS, change the Printer-AMS or Slicer\Device-AMS selection,  then either 
RESYNC - and change presets over the top
Prests first - then SYNC
[image]

Same outcome\result.
[image]

DokuKaefer, 2023-12-15:
Sorry, I haven't read all the entries.
 If I select my own filament, this disappears again. Is this behavior already known?

[image]

OzDruiD, 2023-12-16:
@DokuKaefer 
A few other similar thread  
https://github.com/bambulab/BambuStudio/issues/3111
https://github.com/bambulab/BambuStudio/issues/3087

DanBao-Bambu, 2023-12-18:
@DokuKaefer You can try entering the Filament tab of the printer screen and clicking to enter a detailed material settings page. At this point, the printer will synchronize data from the cloud, and then you will not see "?" when setting up Filament from the Device page in Studio.
Note that the synchronization cloud in the preference settings needs to be turned on.

DokuKaefer, 2023-12-30:
@DanBao-Bambu  

sorry for late reply.
Can you explain in more detail? Do you mean the external printer monitor on the X1C?
If so, I don't see any of my self-created filament presets there.

Another missing point would be the AMS overview.
Here you can't see which vendor is in which slot.
Would it be possible to show the vendor in addition to the filament type?
Example:
[image]

DanBao-Bambu, 2023-12-31:
If you create a custom Filament and still cannot see it on the printer screen, it may have been an error during the creation process. Currently, one of the creation paths that I know of is causing this problem (the printer cannot see the already created Filament): that is, you select the option "Copy current preset", and then select the user preset to create in the drop-down box. Please confirm if you created the new Filament in this way.
Regarding your other requirement, I can bring it up to the relevant personnel, but I suspect there is a reason why they may not display this vendor, and may not be able to display it due to other factors? Anyway, I think your needs are reasonable. So I will discuss this requirement with relevant personnel.