bambulab/BambuStudio #1937: "add pause" not adding pause to the correct layer

moswald14, 2023-06-17:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

CPU: AMD Ryzen 7 2700X
RAM: 32GB
GPU: Nvidia GTX 1080

### Printer

Bambu Lab P1P

### How to reproduce

I think its because I used a variable layer height.
when I slice the file and add a pause gcode using the sliding bar on the right, it adds the pause to the incorrect layer.
in my specific case, it adds the pause to the layer after it. when I add the pause to the layer before it operated like normal. so I cant just add the pause to the layer before.

### Actual results

[image]

### Expected results

The pause should happen on the layer selected.

### Project file & Debug log uploads

[pauseBUG.zip](https://github.com/bambulab/BambuStudio/files/11779696/pauseBUG.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

LiZ-Li-BBL, 2024-01-16:
This issue should have been fixed recently. Try to use the latest version https://github.com/bambulab/BambuStudio/releases/tag/v01.08.03.89

tutek1, 2024-09-23:
I use BambuStudio 1.9.5.51, and this issue still happens. It cost me an 15 hour print :c

lanewei120, 2024-09-23:
if it can not pause at all, it should be another issue

@tutek1 you can report a printer related issue to our support tickets
https://bambulab.com/en/my/support/tickets/create

for this issue, we can close now

closed 2024-09-23 by lanewei120

tutek1, 2024-09-23:
No I can pause just fine. However with variable layer height the pause is not always added to the gcode.

jbruby, 2024-10-01:
This issue is still not fixed as of Bambu Studio 1.9.7.52. When adding a pause to a model using variable layer height, the exported G-code shows the pauses offset by 1 layer. This doesn't always happen, so it seems it may be a rounding issue.

lanewei120, 2024-10-02:
> This issue is still not fixed as of Bambu Studio 1.9.7.52. When adding a pause to a model using variable layer height, the exported G-code shows the pauses offset by 1 layer. This doesn't always happen, so it seems it may be a rounding issue.
> 
> I have attached a 3d benchy bambu studio file I quickly created with variable layer height with a pause set at layer 31. The exported G-code is included which has the pause (M400 U1) at layer 32. [3dBenchy_variable_layer_height_wPause.zip](https://github.com/user-attachments/files/17198805/3dBenchy_variable_layer_height_wPause.zip)

ok, we will follow up

coreyjkelly, 2025-03-20:
I believe I just encountered this same issue in 1.10.2.76. Using adaptive layer height and inserting a pause at layer 75 - the pause shows up at layer 74 in the G-code.

[.3mf and .gcode files](https://github.com/user-attachments/files/19356984/wrong_layer_pause.zip)

[image]

[image]