bambulab/BambuStudio #1940: Build on Arch Linux fails with Target "openvdb_static" links to: Blosc::blosc but the target was not found

ajacocks, 2023-06-19:
### Bambu Studio Version

1.6.2.4 or -current

### Where is the application from?

Built from source by yourself or others

### OS version

Arch Linux x86_64

### Additional system information

```
Memory:
  System RAM: available: 7.76 GiB used: 572.6 MiB (7.2%)
  Array-1: capacity: 8 GiB slots: 1 EC: Multi-bit ECC
  Device-1: DIMM 0 type: RAM size: 8 GiB speed: N/A
CPU:
  Info: 8-core model: Common KVM bits: 64 type: MCP cache: L2: 32 MiB
  Speed (MHz): avg: 1992 min/max: N/A cores: 1: 1992 2: 1992 3: 1992 4: 1992
    5: 1992 6: 1992 7: 1992 8: 1992
Graphics:
  Device-1: driver: bochs-drm v: N/A
  Display: server: X.org v: 1.21.1.8 with: Xwayland v: 23.1.2 driver: X:
    loaded: modesetting gpu: bochs-drm tty: 80x24 resolution: 1280x800
  API: OpenGL Message: GL data unavailable in console and glxinfo missing.
Drives:
  Local Storage: total: 100 GiB used: 75.81 GiB (75.8%)
  ID-1: /dev/sda vendor: QEMU model: HARDDISK size: 100 GiB

$ gcc --version
gcc (GCC) 13.1.1 20230429
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

### Printer

Bambu X1 Carbom w/ AMS

### How to reproduce

1. git clone https://aur.archlinux.org/bambustudio-git.git
```
Cloning into 'bambustudio-git'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 46 (delta 27), reused 34 (delta 18), pack-reused 0
Receiving objects: 100% (46/46), 21.37 KiB | 10.69 MiB/s, done.
Resolving deltas: 100% (27/27), done.
```
2. cd bambustudio-git
3. makepkg -si

### Actual results

-- ----------------------------------------------------
-- ----------- Configuring OpenVDBBinaries ------------
-- ----------------------------------------------------
-- Configuring done (0.0s)
CMake Error at openvdb/CMakeLists.txt:443 (target_link_libraries):
  Target "openvdb_static" links to:

    Blosc::blosc

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

-- Generating done (0.0s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[2]: *** [CMakeFiles/dep_OpenVDB.dir/build.make:96: dep_OpenVDB-prefix/src/dep_OpenVDB-stamp/dep_OpenVDB-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:426: CMakeFiles/dep_OpenVDB.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

### Expected results

Package builds successfully.

### Project file & Debug log uploads

[build.zip](https://github.com/bambulab/BambuStudio/files/11783835/build.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-06-20:
a bit strange, from your log, dep_Blosc has been built successfully
the cmake files have been installed to /home/ajacocks/src/bambustudio-git/src/dep_linux/usr/local/lib/cmake
we need to check why it can not be found by other module

could you upload the files under this directory to have a check also?

Sanaki, 2023-06-21:
Just as a data point, I built the AUR package via pamac without error using GCC-13 after the update on the 19th. I couldn't say why this error represents here and not in pamac's build though.

ajacocks, 2023-06-22:
Here are the build artifacts. Beware, it's a large zip file:

[build.zip](https://www.dropbox.com/sh/dkrx38oy5rk8487/AACmMDPnAzjKYRBCileT7uLQa?dl=0)

lanewei120, 2023-06-25:
I have checked the files generated by building dep_Blosc, can not find any issues.
and I also replaced my cmake files&&libblosc.a to yours
the building can also success

so it seems your files are correct
maybe it is related with your building environment

you can try the original building commands(on our wiki)
and compare the difference to check the reason

currently I can not find a archlinux machine and can not reproduce your issue here

lanewei120, 2023-08-05:
it seems the compile has been finished in another thread, close this one now

closed 2023-08-05 by lanewei120