bambulab/BambuStudio #1942: holes in a model, when using filltool to color some of the parts

madsholme, 2023-06-19:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

bambulab x1c

### How to reproduce

download this model: https://www.printables.com/model/229666-friendly-articulated-slug/files
use the fill tool to fill each "ring" with a color, then slice, and you will see there is holes in the model now. If you dont use the filltool, the holes are not there. 

### Actual results

[image]

### Expected results

a proper model without holes.

### Project file & Debug log uploads

[SlugError.zip](https://github.com/bambulab/BambuStudio/files/11786028/SlugError.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-07-10:
You can use the following method to avoid this problem:
Step 1: Split the model into parts.
[image]
Step 2: Assign corresponding colors to each part. 
[image]

madsholme, 2023-07-10:
Yes that works, and also what i ended up doing. I just thought it would be nice to have it reported as a bug?

zhimin-zeng-bambulab, 2023-07-10:
Thanks for reporting this bug, we will further investigate the root cause.