bambulab/BambuStudio #1944: BambuStudio got an unhandled exception: Calling AppConfig::save() from a worker thread

ferenj-gs, 2023-06-20:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	12th Gen Intel(R) Core(TM) i7-1260P, 2100 Mhz, 12 Core(s), 16 Logical Processor(s)
Installed Physical Memory (RAM)	32,0 GB
Adapter Type	NVIDIA T550 Laptop GPU, NVIDIA compatible
Adapter Type	Intel(R) Iris(R) Xe Graphics Family, Intel Corporation compatible

### Printer

X1-Carbon

### How to reproduce

1. Create App profile under windows Graphics settings for bambu studio to use the dedicated T550 GPU
2. Start Bambu Studio

### Actual results

 Crash happens immediately
[image]

after hitting ok:

[image]

### Expected results

Bambu studio should start normally.

### Project file & Debug log uploads

[debug_network_Tue_Jun_20_10_58_31.zip](https://github.com/bambulab/BambuStudio/files/11798118/debug_network_Tue_Jun_20_10_58_31.zip)

FYI: 0 byte logfile is generated, most probably as this is a startup issue

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-06-25:
[image]

Please create a minidump when this dialog show.
You can create minidump with the following steps:
* Open the Windows Task Manager
* Find and select the Bambu Studio process in it
* Right click the process and choose the 'Create Minidump File' menu item
* Select a folder to save the file

lanewei120, 2023-06-29:
@ferenj-gs 
thanks for reporting this issue

is this issue reproducible?

from log
[image]
it seems the thread_name of BambuStudio is wrong in your environment
it cause the context check failed

and what does this mean "Create App profile under windows Graphics settings for bambu studio to use the dedicated T550 GPU"

lanewei120, 2023-07-27:
> is this issue reproducible?

@CrashTestCoder 

is this issue 100% reproducible?

CrashTestCoder, 2023-07-27:
Will verify when I get back to my computer in the morning. But it persisted through reinstalling the software and multiple reboots. 

Disabling the manual gpu selection fixed it immediately

CrashTestCoder, 2023-07-27:
I just tested on my old personal laptop and it is not reproducible with a Nvidia 940mx. So maybe its just a problem with quadros? Will report back in the morning

CrashTestCoder, 2023-07-27:
Can confirm. 100% reproducible. 
[image]
[image]
[image]

[debug_network_Thu_Jul_27_07_19_04.log](https://github.com/bambulab/BambuStudio/files/12182679/debug_network_Thu_Jul_27_07_19_04.log)

### PC Hardware:
Device name	<REDACTED>
Full device name	<REDACTED>
Processor	11th Gen Intel(R) Core(TM) i5-11400H @ 2.70GHz   2.69 GHz
Installed RAM	32.0 GB (31.7 GB usable)
Device ID	<REDACTED>
Product ID	<REDACTED>
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

[image]
[image]

lanewei120, 2023-07-27:
thanks very much
it should be a similiar issue as https://github.com/prusa3d/PrusaSlicer/issues/9420
we will fix it soon

thanks

lanewei120, 2023-08-05:
already fixed in v1.7.2.51 hotfix version

closed 2023-08-05 by lanewei120