bambulab/BambuStudio #1945: Bottom shell layers should use top shell color/filament

bradMatias, 2023-06-21:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

View layer 15 in the attached pre-sliced 3mf file, the "ALL IN" text bottom shell should be white (or black), but definitely not red.

### Actual results

Colored model:
[image]
Red color visible under text:
[image]
Layer 15 showing incorrect color used for bottom shell:
[image]

### Expected results

The bottom shell color should not choose a color that is both not the text color or the background color.

### Project file & Debug log uploads

[AllIn_PLA_1h33m.zip](https://github.com/bambulab/BambuStudio/files/11813995/AllIn_PLA_1h33m.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-06-28:
> AllIn_PLA_1h33m.zip

This is a 3mf without model mesh, and only has gcode. Besides, the gcode is single filament
[image]
Could you share one 3mf with models so we can reslice with multi color?

bradMatias, 2023-06-28:
Sorry about that! On my end I just tried to open the file and it shows 4 filaments, here is the full project file that you can try slicing
[AllIn_bugged.zip](https://github.com/bambulab/BambuStudio/files/11900696/AllIn_bugged.zip)