bambulab/BambuStudio #1947: Print Sequence by Object + Height Range Modifier (+Colors) = Error/Crash

lmnx, 2023-06-23:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 (x86-64), MacOS Ventura (arm64)

### Printer

X1C + AMS

### How to reproduce

1. Print Sequence: "By object"
2. Add 2 objects
3. Create height range modifiers for one object and modify color once at any layer
5. Slice
6. Get error message or crash

### Actual results

Crash or get error message:

"A fatal error occurred: "Access violation"
Please save your project and restart the application."

### Expected results

Slice normally

### Project file & Debug log uploads

[debug_Fri_Jun_23_14_04_37_1532.zip](https://github.com/bambulab/BambuStudio/files/11845771/debug_Fri_Jun_23_14_04_37_1532.zip)
[Cube.zip](https://github.com/bambulab/BambuStudio/files/11845772/Cube.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2023-06-25:
I have tested with latest code base. The UI can't change color, and also has some other abnormal behaviour. Shown as gif below.
[image]
@bambu123 Please have a check and fix the UI issue first and we will check whether still has crash.

SaltWei, 2023-06-28:
Fixed in internal version. Will be included in next user release version

lmnx, 2023-06-28:
@SaltWei awesome! Waiting to test and will mark this issue as done.

jaddel, 2023-07-05:
Please also have a look a this one, I guess it's directly related.

https://github.com/SoftFever/OrcaSlicer/issues/1272

SaltWei, 2023-07-11:
> Fixed in internal version. Will be included in next user release version

Has be included in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.53. And slicing is normal in my environment.

Could someone help to double check?

lmnx, 2023-07-11:
Thanks, everyone. Looks fixed.

closed 2023-07-11 by lmnx