bambulab/BambuStudio #1948: Missing "Arrow" or "greater than" within sub-menu when changing filaments for objects

maxmax3, 2023-06-23:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 11.7.1

### Printer

Bambu Lab X1 CArbon

### How to reproduce

When switching filaments for objects, you open a context menu to look for the text “change filament" to open a sub-menu to choose the right filament. IMO there's an arrow missing right of the text "change filament", indicating that there's a sub-menu. Would be fine. That's all.

### Actual results

[image]

### Expected results

A text like: "change filament      >"

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-06-25:
It may relate to the old macOS version. I can't reproduce it on the macOS 12/13 version.

bambu123, 2023-07-03:
@maxmax3 Are other sub-menus entrance losing this arrow indicator?

SaltWei, 2024-01-18:
[image]
Looks ok on Windows system

SaltWei, 2024-01-18:
[image]
[image]

Also check on macos and be OK as well.