bambulab/BambuStudio #1950: Device monitoring window causes rendering to stutter in other 2D/3D applications when moving the mouse

skysyk, 2023-06-23:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 5800X3D
RAM: 64 GB
GPU: NVIDIA RTX 4090

### Printer

X1C

### How to reproduce

1. Open Bambu Studio
2. Switch to the Device tab (the video feed does not need to be active)
3. Switch to a different 2D/3D rendering application (e.g. [osu!](https://osu.ppy.sh/))
4. Moving the mouse results in stutters / frametime spikes in the other application

### Actual results

I experience frametime spikes and input lag on the order of once per second when the above steps are followed. In my case, this has happened when playing games such as osu! or Dota 2.

The behavior here is pretty weird and does not occur when Bambu Studio is in any other tab. I do see a 1000 ms timer refresh event in Monitor.cpp, perhaps there is an issue in the way the timer event and wxWidgets are being used that may have negative externalities on other apps?

### Expected results

Bambu Studio does not interfere with other applications running in the background or foreground.

### Project file & Debug log uploads

Project files do not apply.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2023-08-09:
May I ask if this is the download link for OSU https://osu.ppy.sh/home/download

Haidiye00, 2023-08-17:
May I ask if there is any lag phenomenon after turning off the video playback on the device monitoring page

Haidiye00, 2023-09-04:
[image]
like this

EricDimitri, 2024-03-09:
Hi Guys, I experience this same issue, but it is not related to CPU or GPU, it is related, I think, to page faults. Toulsands of pagefaults per second, even when the Bamnbu Studio is not touched for minutes. Page faul ts keep growing and growing, and that looks weird.