bambulab/BambuStudio #1951: Allow users to provide default settings for new project

julie777, 2023-06-23:
**Is your feature request related to a problem? Please describe.**

I really do not like Bambu Studio using settings from the current project when I create a new project. **That is never what I want**, especially if I have customized individual settings from their profile values.

For many users the default would 
- printer - the only printer nozzle combination that they have
- filament - Generic PLA
- process - standard .4 nozzle.

For me it would be 
- printer - P1P .4 fast
- filament - Sunlu PLA+
- process - my .4 standard

**Describe the solution you'd like**

By default, Bambu Studio should use a reasonable default for all three profile types when creating a new project.

There should be a preferences setting that allows the user to set the default for each of the 3 profile kinds to what they want to use for new projects.

MadMax389, 2023-07-03:
My workaround is to create a blank project with a cube primitive on it with the settings I want, one for my X1C and one for my P1P. I open the blank first, then create a new project. It keeps all the settings from the blank including profiles, colors and flush volumes. Only takes a few extra seconds.

julie777, 2023-12-13:
I would like to be able to configure the presets to be used for any new projects. Just add it to settings/preferences.