bambulab/BambuStudio #1952: Crashing while slicing plate

inktomi, 2023-06-24:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.4.1 (22F82)

### Additional system information

Apple M1 Max
32 GB Ram

### Printer

Bambu Labs X1C

### How to reproduce

Attempt to slice second plate.

### Actual results

Bambu Studio crashes. 

### Expected results

Plate is sliced so that it can be printed

### Project file & Debug log uploads

[Crashing Print Plate.3mf.zip](https://github.com/bambulab/BambuStudio/files/11858089/Crashing.Print.Plate.3mf.zip)
[debug_Sat_Jun_24_13_23_47_7536.log.0.zip](https://github.com/bambulab/BambuStudio/files/11858090/debug_Sat_Jun_24_13_23_47_7536.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-06-25:
Could you also help to find the crash report in the "console" application
and upload here?

https://support.apple.com/en-gb/guide/console/cnsl664be99a/mac

inktomi, 2023-06-25:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [7440]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.06.02.04 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-06-24 13:23:34.0938 -0700
OS Version:            macOS 13.4.1 (22F82)
Report Version:        12
Anonymous UUID:        AA78B9B4-FFC5-03BC-2172-1AA8F834C210

Sleep/Wake UUID:       1D5872B8-ACF7-489D-9C49-37509D2E551B

Time Awake Since Boot: 25000 seconds
Time Since Wake:       1339 seconds

System Integrity Protection: enabled

Crashed Thread:        16  slic3r_tbb_4

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Codes:       0x0000000000000001, 0x0000000000000018

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [7440]

VM Region Info: 0x18 is not in any region.  Bytes before following region: 105553518919656
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1a5cbff14 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x1a5cd2240 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x1a5cc8b78 mach_msg_overwrite + 604
3   libsystem_kernel.dylib        	       0x1a5cc0290 mach_msg + 24
4   CoreFoundation                	       0x1a5dde7e4 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x1a5ddd0c4 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x1a5ddc4b8 CFRunLoopRunSpecific + 612
7   HIToolbox                     	       0x1af626c40 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1af626a7c ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x1af6267d4 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x1a8ffdd44 _DPSNextEvent + 636
11  AppKit                        	       0x1a8ffcee0 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
12  AppKit                        	       0x1a8ff1344 -[NSApplication run] + 464
13  BambuStudio                   	       0x104eeaba0 wxGUIEventLoop::OSXDoRun() + 112
14  BambuStudio                   	       0x104fc374c wxCFEventLoop::DoRun() + 40
15  BambuStudio                   	       0x104f42980 wxEventLoopBase::Run() + 132
16  BambuStudio                   	       0x104f1d810 wxAppConsoleBase::MainLoop() + 120
17  BambuStudio                   	       0x104eb0e78 wxApp::OnRun() + 36
18  BambuStudio                   	       0x104f7045c wxEntry(int&, wchar_t**) + 148
19  BambuStudio                   	       0x103174840 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 172
20  BambuStudio                   	       0x10275386c Slic3r::CLI::run(int, char**) + 21740
21  BambuStudio                   	       0x102772014 main + 172
22  dyld                          	       0x1a59a7f28 start + 2236

Thread 1:
0   libsystem_pthread.dylib       	       0x1a5cfad8c start_wqthread + 0

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	       0x1a5cbff14 mach_msg2_trap + 8
[... 646 lines cut ...]
    {
      "rolloutId" : "6425c75e4327780c10cc4252",
      "factorPackIds" : {
        "SIRI_HOME_AUTOMATION_INTENT_SELECTION_CACHE" : "642600a457e7664b1698eb32"
      },
      "deploymentId" : 240000004
    },
    {
      "rolloutId" : "62699e1ec1ff2978b47f6c3b",
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "631f72d1de559130376df1fb"
      },
      "deploymentId" : 240000023
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c28e4ee6-1b08-4f90-8e05-2809e78310a3",
      "experimentId" : "6317d2003d24842ff850182a",
      "deploymentId" : 400000013
    }
  ]
}
}
```

SaltWei, 2023-06-25:
@bambu123 这个crash的问题 @zhimin-zeng-bambulab 已经修复过了，我在最新code base不复现这个crash 的bug。
不过切片第二盘会卡在这里卡一会。请椰子看一下。
[image]

inktomi, 2023-06-25:
If I tweaked the angle that the objects are at, the plate slices. But the original angle, it crashes.

SaltWei, 2023-06-25:
> If I tweaked the angle that the objects are at, the plate slices. But the original angle, it crashes.

The 3mf attached on the tickets is 100% crashed on your environment, right ?

SaltWei, 2023-06-25:
We have one fix in internal version and I have tested the attached 3mf file, and cannot reproduce the crash. The fix may work.

Manfredo1952, 2023-07-02:
@SaltWei   when will the nex user version be made available then ?

lanewei120, 2023-07-03:
we are preparing this version
and it will be released as soon as some functions stable and bug fixed

however we do not a fixed date yet

Manfredo1952, 2023-07-03:
@lanewei120    If you need beta tester, let me know please.     Regards, Manfred

SaltWei, 2024-01-16:
Does this issue still exist on latest version?
Close the issue at the moment, Reopen if still exists

closed 2024-01-16 by SaltWei