bambulab/BambuStudio #1953: Make all solid infill monotonic when Top surface pattern is Monotonic Line.

gary-wlg, 2023-06-24:
The top surface can have poor quality when using Monotonic Lines as the previous layers show through since it is not done as monotonic.

If you select Monotonic as the Top source pattern, then all solid layers below it are also printed as Monotonic. This should also be done for Monotonic Lines, as right now if you select Monotonic Lines, the solid layers below it are not printed Monotonic.

Visible scaring shows through on dark filaments.

[image]

QingZhangBambu, 2023-07-03:
good point

Gask44, 2026-08-29:
I agree with this request, but I think it could be taken one step further.
Instead of applying Top Surface settings only to the final visible layer, Bambu Studio could allow the user to choose how many of the final solid layers use the Top Surface quality settings — for example 2 or 3 layers.
Ideally, the direction could alternate between them:
Layer -2: 0°
Layer -1: 90°
Top layer: 0°
This would give the final Top Surface a much more uniform foundation. The layers directly underneath have a significant effect on the appearance of the final surface, especially on large flat parts.
This is not the same as simply increasing Top Shell Layers. The idea is to let the last 2–3 solid layers actually use the Top Surface speed, acceleration, line width and other relevant quality parameters.
I think this could noticeably improve top surface quality without requiring ironing.