bambulab/BambuStudio #1954: Have the ability to select a different z-hop type for the first layers of small islands

gary-wlg, 2023-06-24:
Background
When printing the bottom layer of a model that has de-bossed text on it, you end up with some very small "island" extrusions, (think of the centre of an "O" or an "R"). These small extrusions can become detached with the default Spiral Z-Hop type, since it is quite aggressive.

Perhaps the slicer could be smart enough to find these first layer islands and change the z-hop type to Normal if it is set as spiral.  Another option could be  to allow you select a different z-hop type using a modifier, or a layer height modifier.

SaltWei, 2023-07-12:
[image]

SaltWei, 2023-07-12:
> Another option could be to allow you select a different z-hop type using a modifier, or a layer height modifier.

You can use modifier or height range to set filament. And the filament can be set to other lift type. Shown as above picture.