bambulab/BambuStudio #1955: slicing 3mf file is not generating an accurate print compared to the model or the STL version of the same file

tsdexter, 2023-06-25:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

OSX 13.3

### Additional system information

2.2 GHz 6-Core Intel Core i7
Radeon Pro 560X 4 GB
Intel UHD Graphics 630 1536 MB
32 GB 2400 MHz DDR4

### Printer

X1C with AMS

### How to reproduce

1. Open the included file which has a 3MF model and STL model, both looking like: 
[image]

2. Slice plate 1 which has the 3MF version, which results in having weird issues with the `O`, the dot over the top of the `i` is entirely missing, and the lower corners of the `i` and `e` have odd indents: 
[image]

3. Slice plate 2 which is the STL version, this one was generated by right-clicking the 3MF model and choosing `Export as STL`, this results in a properly sliced model: 
[image]
 

### Actual results

3MF version of the file does not slice correctly

### Expected results

3MF version should slice the same as the STL version

### Project file & Debug log uploads

[zowie-bambustudio.3mf.zip](https://github.com/bambulab/BambuStudio/files/11858549/zowie-bambustudio.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tsdexter, 2023-06-25:
note: I need to use the 3mf version as this is part of a larger multipart model which cannot be saved in STL format with all of the parts in the same file

SaltWei, 2023-06-25:
[image]
[image]
These two models have different model error, so have different sliced result.

SaltWei, 2023-06-25:
Repair the models and both of them can be sliced rightly. But there has no model repairing function on mac system. Model repairing can only work on Windows   @lanewei120

tsdexter, 2023-06-25:
@SaltWei any idea when model fixing is coming to mac?

lanewei120, 2023-06-29:
> @SaltWei any idea when model fixing is coming to mac?

hello

model fixing is a bit complex, we use the windows sdk on windows 
however it can not be used on mac

we also have tried some developments,
but can not achieve a good result

it is a long time work, and we can not give a fixed schedule currently