bambulab/BambuStudio #1957: Turn off P1P's LED after print finishes

capsel22, 2023-06-25:
**Is your feature request related to a problem? Please describe.**
The LED turns on automatically when the print starts but doesn't turn off when the print finishes.

**Describe the solution you'd like**
The LED to turn off after print finishes.

**Describe alternatives you've considered**
Using third party home assistant integration to trigger on state change from printing to idle and turn off light off

tkaravou, 2023-06-25:
I second this, there are ways to do this via GCode but this seems like a valid setting to have for when a print starts / stops.

bjoernmayer, 2023-06-25:
>  there are ways to do this via GCode

Please elaborate. What GCode can be used?

capsel22, 2023-06-25:
> > there are ways to do this via GCode
> 
> Please elaborate. What GCode can be used?

I second that question

HOLLY-DION, 2023-06-30:
I think the vast majority of the time I do not need LED lights. Prefer to have it off by default.

alienwaredream, 2023-07-09:
> > there are ways to do this via GCode
> 
> Please elaborate. What GCode can be used?

Agree

Str00pwafel, 2023-07-15:
Would love to have this, or the gcode to add this myself.

VolodymyrSemenov, 2023-07-21:
Agreed

Faasteer, 2023-07-28:
I would love to have this feature on the P1S as well

bjoernmayer, 2023-08-09:
Is this fixed? I printed today with the new 1.04 firmware and the flash turned off after the print.

Str00pwafel, 2023-08-09:
> Is this fixed? I printed today with the new 1.04 firmware and the flash turned off after the print.

I dont think so. Even so, when I turned off the LED and started a new print, it didn't turn on...
Seems to be worse than before the new firmware.

dkbay, 2023-11-12:
> > there are ways to do this via GCode
> 
> Please elaborate. What GCode can be used?

source: https://forum.bambulab.com/t/bambu-lab-x1-specific-g-code/666

| Printhead gcode | Description                       |
|-----------------|-----------------------------------|
| M960 S5 P0      | Bambu Lab toolhead logo light off |
| M960 S5 P1      | Bambu Lab toolhead logo light on  |

darrylluther, 2023-11-12:
1. The issue is with the LED lamp, not the tool head logo LED. 

2. The P1P does not have a tool head logo LED.

ctiemann, 2023-12-09:
I would also ask that the function be implemented. It is annoying that the LED for the printing plate stays on after printing. The whole room is illuminated at night. Please implement or provide a GCode.

cmodyssey, 2023-12-28:
I would really like to have options such as:

During print:
LED Illumination: on or off
Tool Head LED : on or off

After print:
LED Illumination: left it current state or turned off
Tool Head LED : left it current state or turned off

This should cover all varients of what different people would want.

miru8112, 2024-01-02:
I came here to suggest this for the P1S as well. 
I drint longer prints during the night. And it drives me crazy to know that there is an unnecessary light burning for 5+ hrs after my print is done while everybody is asleep. Since I want the timelapse to see where things went wrong, WHEN they go wrong, I also can't just turn the lights off... 

Please add the possibility to either turn the lights off after the print or - better- to set a timer, like 15 minutes after the print finished... got to sleep, meaning: Lights off, Standby, etc. 
thank you

miru8112, 2024-01-02:
> > there are ways to do this via GCode
> 
> Please elaborate. What GCode can be used?

We can't. 
AFAIK Bambu uses a proprietary GCODE in many cases. 
Meaning that the "Normal" commands that would work with an Ender or Prusio will most likely not work on the Bambu Printers as they use differend commands / codes.

PandaBalls, 2024-01-04:
`M960 S5 P0 ;turn off logo lamp`
Doesn't works...
+1 Turn off LED after printing.

MonkaKokosowa, 2024-01-04:
Also this would be a great feature on A1 series

capsel22, 2024-01-04:
the only way to reliably turn the LED off for me was to make an automation in Home Assistant.

Phil-Barker, 2024-02-16:
> the only way to reliably turn the LED off for me was to make an automation in Home Assistant.

Thank you for this!

I've been frustrated that every time I print on my P1S the chamber light is then on forever... I mean it's not a lot of electricity but it's completely unecessary

Never thought of using HASS - created an automation, on print complete, waits 5 minutes, turns off light. Works perfectly 😁

3DJupp, 2024-03-04:
Would love to see this as an official firmware setting. Shouldn't be to hard to implement.

Macelovic, 2024-04-20:
This is actually the only thing that bothers me on my printer...

M1S2, 2024-04-20:
Same problem with my A1. Camera LED ist always turned on automaticaly at the start of a print but never turned off.
"M960 S5 P0" command doesn't work.
Please provide either a gcode to control the LED or don't automaticaly turn the LED on! Let the user configure their preferred behaviour.

zwamdurkel, 2024-05-19:
It's really painful to see that this has been an issue for such a long time. I often forget that the LED light stays on after a print.

Why would I have to turn it off myself? I did not turn it on. If the printer turns on the light when it starts a print, it should also turn it off after a print!

chiwon575, 2024-05-21:
i would also like this feature enabled.  just add a simple control.  lamp on or off when print begins.   lamp on or off when print ends.  what is so complicated

TUCAOEVER, 2024-06-21:
Still not fixing...

efab2022, 2024-07-03:
+1 please

tonycstech, 2024-07-08:
Any update on that ?
Need to turn off LED inside the chamber when print is done.
This can be used to trigger a relay to disconnect the printer from the MAIN power so it does not stay ON all the time feeding network traffic to Chinese servers.

bhuu3, 2024-07-30:
Great idea +1

pedro-vale, 2024-08-03:
If I am not mistaken, it seems to be "just" making a call with `MachineObject::command_set_chamber_light` [method](https://github.com/bambulab/BambuStudio/blob/d59cb73a8d0893fb8c00cb7ecc01d59130bf2fba/src/slic3r/GUI/DeviceManager.cpp#L1931) whenever the printer enters a finish state using a callback function. Just point us the finish state callback function then we can add this method there by ourselves.

tonycstech, 2024-08-03:
> If I am not mistaken, it seems to be "just" making a call with `MachineObject::command_set_chamber_light` [...]

And how exactly can i do that ?
Can i send this to a machine as a GCODE with the print ?

TUCAOEVER, 2024-08-05:
> If I am not mistaken, it seems to be "just" making a call with `MachineObject::command_set_chamber_light` [...]

This could be related to the ams light function not the machine led.

tonycstech, 2024-08-05:
> This could be related to the ams light function not the machine led.

Either way, how do you go about executing it via GCODE ?

pedro-vale, 2024-08-05:
Maybe it shouldn't be something sent via `gcode`. I think this is a machine behaviour related thing.

I was skimming through the code and found these two methods that provide printing status:
- https://github.com/bambulab/BambuStudio/blob/d59cb73a8d0893fb8c00cb7ecc01d59130bf2fba/src/slic3r/GUI/DeviceManager.cpp#L2498
- https://github.com/bambulab/BambuStudio/blob/d59cb73a8d0893fb8c00cb7ecc01d59130bf2fba/src/slic3r/GUI/DeviceManager.cpp#L2526

tonycstech, 2024-08-05:
Maybe it should

InterstellarBeing, 2024-08-22:
wow, over a year and no appropriate solution :(

mattold7, 2024-08-31:
+1 by me as well, I would really like this feature

ctiemann, 2024-08-31:
It's really bad. Many users have been asking for this for over a year and nothing has happened.
Many updates have been released for the P1P since then, including in the Bambu-Lab software. Is it so difficult to integrate this option or are the developers responsible not interested?

GeorgeMA-Strong, 2024-09-22:
STILL WAITING! I'm really tired from turning off light before night by myself!

adripo, 2024-09-24:
Is it possible to use what https://github.com/TFyre/bambu-farm has implemented https://github.com/TFyre/bambu-farm/blob/main/bambu/src/main/java/com/tfyre/bambu/printer/BambuPrinterImpl.java#L305 and trigger it when the print progress is 100%?

altimac, 2024-09-29:
Crazy how many people are asking for such as simple and basic feature and ... it's still open.
I really would like the LED to be automatically OFF after print. And please give us some settings (possibly even a timer after print is done)

RobMcA, 2024-10-24:
I'm looking for this feature as well. Ideally once the print is done, the light goes back to its state before the print started. So if it was on, it stays on, if it was off, it turns off. It's very annoying to have to manually turn it off after every print.

disconn3ct, 2024-10-29:
Workarounds:
If you use mqtt directly, you can force it off easily enough: https://github.com/davglass/OpenBambuAPI/blob/main/mqtt.md#systemledctrl Probably a simple `mosquitto_pub` command could do it.
If home assistant is your jam, https://github.com/WolfwithSword/Bambu-HomeAssistant-Flows takes it through as a `light` entity (along with a ton of other features)

InterstellarBeing, 2024-10-29:
> Workarounds:
>
> If you use mqtt directly, you can force it off easily enough: https://github.com/davglass/OpenBambuAPI/blob/main/mqtt.md#systemledctrl Probably a simple `mosquitto_pub` command could [...]

Thanks! Will have a look on mqtt - never worked with that

zwamdurkel, 2024-10-29:
I just realized this now, but if this is a feature implemented in Bambu Studio, it won't really fix the problem because then the app needs to stay on. This should actually be a firmware update on the printer.

tho-myr, 2024-12-07:
Hey i found this video on this topic which might be a workaround. Haven't tried it yet but it seems to be possible to apply a lamp turn off command after printing is finished :)

https://youtu.be/UrWaExvoCcc?si=tLDBub2MfjjgBld1

3DJupp, 2024-12-07:
> Hey i found this video on this topic which might be a workaround. 

You're referring to a gcode which applies to the X-Series. 
For the P-Series bambu states it's addressed in the R&D Team
[image]

tonycstech, 2024-12-07:
it does not work.
It may turn off LOGO light but not chamber light

Roycinger, 2024-12-08:
Seems to be normal that Github issues are completely ignored by "R&D".
Nevertheless, this issue has be open for almost 1 1/2 years, so my assumption above might be correct.

@BambuLab, what do you have an issue section even for?

Ic3n, 2024-12-11:
ok my ticket
[image]
[image]

tho-myr, 2024-12-22:
hey guys, i found a pretty easy solution for everyone who is using home assistant. there is a great bambulab integration which can be installed via the hacs (home assistant community store) and then you can add your printer as device to home assistant. 

with the printer as device in home assistant you can create a really simple automation that turns the chamber light off after print finish/error/cancel. it's super easy to setup and works with my p1p very reliable 🥳

tonycstech, 2024-12-22:
No. There needs to be a way to turn it off via gcode.

Smutchings, 2025-03-26:
+ 1 to adding this feature to the P1x firmware

a-dam-berger, 2025-04-21:
Come on Bambu, it is about time that you added the ability to toggle this in the GCode... it has been 1.5 years since this issue was raised.

tonycstech, 2025-08-04:
Bump all day long.
China does not care what we want

gonpostal, 2025-12-01:
I can't believe this hasn't been fixed yet. IMO, it would be fine if the light shut off when the printer went into sleep mode; seems like it would be a pretty trivial thing to add to a firmware update.

Roycinger, 2025-12-01:
> I can't believe this hasn't been fixed yet. IMO, it would be fine if the light shut off when the printer went into sleep mode; seems like it would be a pretty trivial thing to add to a firmware update.

Let's be realistic.
BambuLabs ignores this Github anyways and we all know this will never be implemented.
I really don't get why they have a Github issue section at all.

Martan404, 2025-12-31:
I would like to see this feature for my P1S.

There is no reason for the machine to keep the lights on when the machine goes idle! This should be an easy fix for Bambu Lab. 

Just add another option bellow light toggle; `If idle, turn off light!` It should be default behavior.

We shouldn't have to mess with G-code and create a custom machine profile to turn the lights off...

Please fix this!

Skye-Bambulab, 2026-11-16:
Hello, thank you for your feedback. We understand the inconvenience caused by the chamber/camera LEDs remaining on after print completion, and we acknowledge the community's need for configurable light settings.

To address this, we have decided on the following approach:

1. Feature Implementation: We plan to introduce an interfacefor chamber LED control via the M960 G-code command. This will allow users to customize their lighting preferences by adding specific commands to their "End G-code," enabling the lights to either switch off or remain in their previous state upon print completion.

2. Release Timeline: Support for this feature on the A/P series willbe rolled out gradually, aligned with our upcoming firmware releaase schedule.

We will keep this issue as a tracked feature request until the implementation is complete. Thank you for your patience and for
helping us improve the user experience.

alienwaredream, 2026-11-16:
> Hello, thank you for your feedback. We understand the inconvenience caused by the chamber/camera LEDs remaining on after print completion, and we acknowledge the community's need for configurable [...]

Wow, I didn’t expect the official team to fix this issue after so many years. Thank you so much! 🙏

[2 more replies]