bambulab/BambuStudio #1958: Change filament does not work correctly on P1P without AMS

julie777, 2023-06-25:
### Bambu Studio Version

1.6.2.3

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1P

### How to reproduce

1. create new project and add an object
2. slice the object
3. move the layer slider to some middle layer
4. right click on the + and choose change color (note that choose color is not available unless a second filament has been added to the filament section of settings.
5. re-slice the model
6. print the model

### Actual results

the model will be printed all in one color, as will the purge block

### Expected results

There are multiple problems. 

Adding colors to the project does not really make much sense without an AMS, but I can see how it would be useful to document the filament colors. However, if the printer does not have an AMS it should not be used to generate gcode.

Change color should always be available in the layer slicer context menu. It is valid to change colors with or without an AMS. Pause at layer can be used as a work-around currently, but is not a good solution.

If there is no AMS, then the sliced model should not have code for a purge block. Instead, it should pause at the correct layer, notify the user to change filament, and start the printer unload and load filament sequence. 

After noticing the "Change Filament Now" notification, the user should go to the print and the screen should prompt the user through the process.
- yes I want to change filament
- yes the filament has been unloaded
- load the new filament
- yes the filament is extruding the correct color
- printer then continues printing

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-06-28:
> 4\. note that choose color is not available unless a second filament has been added to the filament section of settings

The filament section of prepare page doesn't means filaments in AMS. It means the filament used by this 3MF project.

paoloMain, 2023-11-12:
The issue is still present on 1.7.7.89.
Adding a "change filament" pin in the preview page doesn't actually stops the print and the object is all printed with the first material.

SaltWei, 2023-11-12:
> Adding a "change filament" pin in the preview page doesn't actually stops the print and the object is all printed with the first material.

The changing filament is for AMS.
I think you should add pause and change filament manually without AMS

paoloMain, 2023-11-12:
> The changing filament is for AMS. I think you should add pause and change filament manually without AMS
Thanks for the answer :-) but what if I want to print with different materials (pla-petg)? Do I have to manually change all the settings from the printer screen?

brandonkish, 2023-12-07:
> > The changing filament is for AMS. I think you should add pause and change filament manually without AMS
> 
> Thanks for the answer :-) but what if I want to print with different materials (pla-petg)? Do I have to manually change all the settings from the printer screen?

This is a big issue with the pause. If you go from something like PLA to TPU, you can't change the print settings for the new filament i.e. Speed, Acceleration, etc...

Also, any settings changed during the pause are overwritten when the print resumes, the previous settings are restored. So you have to try to change the settings while it is already printing.

It would be nice to implement an optional pause in the "Change Filament" so that those of us using filaments like TPU can manually change it out.

IanVS, 2024-01-03:
I was also confused why the printer just blew past my filament change that I had set up in the slicer, and continued printing with the original color.  What is the point of the slicer creating a purge block and showing me this readout with "Filament change times" if it's not going to actually let me change the filament?  

[image]

I think that by default the printer should add an M600 when a filament change is instructed in the slicer, unless the AMS is used and then it can do whatever it is it's doing now.

GuzziRaz, 2024-01-24:
I was hit with this bug on the X1C now. The model starts with 8 layers of TPU 95A and the rest is PLA. I inserted a Change Filament at layer 9. I loaded the TPU from external spool and started the print with AMS deselected.

Expected results: It should pause at the filament change, letting me manually switch over to a filament in the AMS, or at least change the external spool to the PLA one.

Actual results: No pause, nothing. It continued using the TPU95A but with PLA settings so ended up in caos (thankfully caught by the spaghetti detection).

So I wanted to insert a pause as well, at the same layer height as the filament change, but that doesn't seem to be possible. I am right now trying the same but with `M400 U1` at the very top of the Change Filament G-code. Hopefully that will work but it has minor flaws (that pause triggers at start of print as well, I just resumed it).

This really should be enhanced to do what the user expects. The logic is very simple: If AMS is not in use, there should be a pause for user intervention.

You may think of this as an enhancement request and "function as design" but in a user experience perspective, it's definitely a bug.

IanVS, 2024-01-24:
I found this custom gcode for filament change which isn't perfect (the flap isn't closed while purging, so the poops are stringy), but does allow changing out the filament without an AMS.  https://gist.github.com/adripo/4d0df406ea86de9d1a52f09dab318538.

DusterTheFirst, 2024-03-08:
Lacking this feature is unacceptable. The printers have the functionality to change filaments, why are end users unable to utilize it if they do not have an AMS or filament that supports the AMS.

Chrissi2812, 2025-07-18:
Seems like it finally happened 🤯

According to release Notes: https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60
> External Spool Support Without AMS
> 
> You can now use multiple filaments on a single external spool slot, even without an AMS unit, thanks to new support for manual filament changes during a print. This is especially useful for multi-color or multi-material prints when AMS is not available.
> How It Works:
> 
>     In the Print Job dialog, select the new option to enable external spool swapping (requires compatible firmware).
>     During the print, the printer will pause and prompt you when it’s time to switch filaments.
>     You can then manually change the spool and continue printing.