bambulab/BambuStudio #1960: Add Russian language

stasyan68, 2023-06-25:
[ru.zip](https://github.com/bambulab/BambuStudio/files/11860828/ru.zip)

AndylgTom, 2023-07-06:
@stasyan68 
1. First they must fix the text display problem https://github.com/bambulab/BambuStudio/issues/1707 .
2. The translation that you posted is is very bad, better delete it and take it from the topic that I posted above. It is a test, not final.

1. Сначала они должны пофиксить проблему отображения текста, что описана здесь https://github.com/bambulab/BambuStudio/issues/1707 .
2. Перевод что вы выложили, очень кривой, лучше удали его и возьми из той темы что выше я выкладывал. Он тестовый, не окончательный.

iStarComLab, 2023-07-18:
> @stasyan68
> 
> 1. First they must fix the text display problem [Wrong encoding - cyrillic #1707](https://github.com/bambulab/BambuStudio/issues/1707) .
> 2. The translation that you posted is is very bad, better delete it and take it from the topic that I posted above. It is a test, not final.
> 
> на русском
> 
> 1. Сначала они должны пофиксить проблему отображения текста, что описана здесь [Wrong encoding - cyrillic #1707](https://github.com/bambulab/BambuStudio/issues/1707) .
> 2. Перевод что вы выложили, очень кривой, лучше удали его и возьми из той темы что выше я выкладывал. Он тестовый, не окончательный.

Проверил твой перевод))) От стасяна68 точнее и аккуратнее, свой не показывай ни кому лучше!

AndylgTom, 2023-07-18:
> Проверил твой перевод))) От стасяна68 точнее и аккуратнее, свой не показывай ни кому лучше!

Ну-ну... не понятно кто говорит. Ты наверно попутал. Вот скрин примера сравнение перевода, то что нашел за минуту. И как такое? 
[image]

Во первых, перевод не окончателен как я и написал, ибо проверить всё не представляется возможности из-за глюка в кодировке. 
Во вторых, видно что он подглядывал в перевод Пруши, основным автором которого являюсь я, но это ему не очень помогло.

iStarComLab, 2023-07-24:
>>>перевод Пруши, основным автором которого являюсь я, но это ему не очень помогло.
Это хорошо что Вы автор, но баг известный https://github.com/SoftFever/OrcaSlicer/issues/757
И походу пока разработчик не исправит кодировку - перевод будет кастрированный.
Я дико извиняюсь что мог задеть Вас но не в коем случае не подумайте ничего нехорошего, просто в таком виде переводы не выкладывают от сюда и такая реакция была с моей стороны. Допилите перевод а мы оценим по достоинству!

iStarComLab, 2023-07-26:
туда же https://github.com/SoftFever/OrcaSlicer/discussions/600

lanewei120, 2023-08-05:
@StoneLiBambulab please followup this requirement

StoneLiBambu, 2024-01-29:
added in 1.8， closed

closed 2024-01-29 by StoneLiBambu