bambulab/BambuStudio #1962: Bambu Studio crashes when adding a part whose STL file is locked externally

logiclrd, 2023-06-27:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 1909

### Printer

Bambu Lab X1 Carbon + AMS

### How to reproduce

1. Import an object into Bambu Studio.
2. Design a part in another application, e.g. OpenSCAD. The design should be such that it takes some time to save the exported STL file.
3. Begin saving the STL file in that application.
4. While it is still in the process of saving, attempt to add the new STL file to the project in Bambu Studio as a new part in the object created in step 1.

### Actual results

Bambu Studio immediately crashes and exits.

### Expected results

Bambu Studio should gracefully detect that the file cannot be read and notify the user. Note that it does present an error message when importing a new _object_ from a file that can't be read; the crash only happens when adding a _part_ from a file that can't be read.

### Project file & Debug log uploads

There is no project file to upload, because Bambu Studio crashes abruptly before any project has been generated/saved.

Log files:
[crashlog.zip](https://github.com/bambulab/BambuStudio/files/11877012/crashlog.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-06-30:
I can't reproduce it. Could you help debug?

* Download the dump.txt attached, and rename it to dump.reg
* Double-click the downloaded file to import some registries
* Start Bambu Studio and reproduce the problem.
* Open the folder '%APPDATA%\BambuStdio\log' where the minidump is created
* Send back the DMP files in it

[dump.txt](https://github.com/bambulab/BambuStudio/files/11914127/dump.txt)

bambu123, 2023-12-26:
We will close this issue because there has been no reply for a long time. Please use the latest version. If there is still the same issue, please create a new issue ticket.

closed 2023-12-26 by bambu123