bambulab/BambuStudio #1963: CLI process.json not work

NathaelB, 2023-06-27:
### Bambu Studio Version

1.6.2

### Where is the application from?

Bambu Lab github releases

### OS version

Debian 5.10.162-1 x86_64

### Printer

Bambu Lab x1 Carbon

### How to reproduce

`$ git clone git@github.com:bambulab/BambuStudio.git`
`$ cd ./BambuStudio`
`$ ./DockerBuild.sh`
`$ ./DockerRun.sh`

**IN CONTAINER**
`$ ./bambu-studio --debug 5 --curr-bed-type "Cool Plate" --load-settings "machine.json;process.json" --load-filaments "filament.json" -slice 0 --export-settings settings.json --export-3mf ouput.3mf input.stl`

### Actual results

When executing the command, we want to use these different configuration files:
- `machine.json`
```json
{
    "type": "machine",
    "setting_id": "GM001",
    "name": "Bambu Lab X1 Carbon 0.4 nozzle",
    "from": "system",
    "instantiation": "true",
    "inherits": "fdm_bbl_3dp_001_common",
    "printer_model": "Bambu Lab X1 Carbon",
    "default_filament_profile": [
        "Bambu PLA Basic @BBL X1"
    ],
    "default_print_profile": "0.20mm Standard @BBL X1C",
    "nozzle_diameter": [
        "0.4"
    ],
    "extruder_offset": [
        "0x2"
    ],
    "bed_exclude_area": [
      "0x0",
      "18x0",
      "18x28",
      "0x28"
    ],
    "extruder_clearance_max_radius": "56",
    "machine_pause_gcode": "M400 U1\n",
    "machine_start_gcode": "LONG MESSAGE",
    "scan_first_layer": "0",
    "machine_load_filament_time": "17",
    "machine_unload_filament_time": "16",
    "nozzle_type": "stainless_steel",
    "nozzle_hrc": "20",
    "auxiliary_fan": "0",
    "upward_compatible_machine":[
        "Bambu Lab X1 0.4 nozzle",
        "Bambu Lab X1 Carbon 0.4 nozzle"
    ]
  }
```
[... 533 lines cut ...]
        "2"
    ],
    "wipe_tower_no_sparse_layers": "0",
    "wipe_tower_rotation_angle": "0",
    "wipe_tower_x": [
        "155"
    ],
    "wipe_tower_y": [
        "220"
    ],
    "xy_contour_compensation": "0",
    "xy_hole_compensation": "0",
    "z_hop": [
        "0.4"
    ],
    "z_hop_types": [
        "Spiral Lift"
    ]
}
```

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-06-28:
hi @NathaelB 

it is just because the process.json has inherits, which includes lots of settings
[image]

from our gui app, the app will extract the values from inherits
and from the cli program, it doesn't include this logic

you need to export the full json file by some way
one method is to use bambustudio's "export configs"
[image]

just save a user config based on the system preset which you want to export
and then export it

NathaelB, 2023-06-28:
So I exported my config and obtained this file :
`process.json`
```json
{
    "bottom_shell_layers": "3",
    "bottom_shell_thickness": "0",
    "bottom_surface_pattern": "monotonic",
    "bridge_angle": "0",
    "bridge_flow": "1",
    "bridge_no_support": "0",
    "bridge_speed": "50",
    "brim_object_gap": "0.1",
    "brim_type": "auto_brim",
    "brim_width": "5",
    "compatible_printers": [
        "Bambu Lab X1 Carbon 0.4 nozzle",
        "Bambu Lab X1 0.4 nozzle"
    ],
    "compatible_printers_condition": "",
    "default_acceleration": "10000",
    "detect_narrow_internal_solid_infill": "1",
    "detect_overhang_wall": "1",
    "detect_thin_wall": "0",
    "draft_shield": "disabled",
    "elefant_foot_compensation": "0.15",
    "enable_arc_fitting": "1",
    "enable_overhang_speed": "1",
    "enable_prime_tower": "1",
    "enable_support": "0",
    "enforce_support_layers": "0",
    "ensure_vertical_shell_thickness": "1",
    "filename_format": "{input_filename_base}_{filament_type[0]}_{print_time}.gcode",
    "flush_into_infill": "0",
    "flush_into_objects": "0",
    "flush_into_support": "1",
    "from": "",
    "fuzzy_skin": "none",
    "fuzzy_skin_point_distance": "0.8",
    "fuzzy_skin_thickness": "0.3",
    "gap_infill_speed": "250",
    "gcode_add_line_number": "0",
    "independent_support_layer_height": "1",
    "infill_combination": "0",
    "infill_direction": "45",
    "infill_wall_overlap": "15%",
    "inherits": "0.20mm Standard @BBL X1C",
    "initial_layer_acceleration": "500",
    "initial_layer_infill_speed": "105",
    "initial_layer_line_width": "0.5",
    "initial_layer_print_height": "0.2",
    "initial_layer_speed": "50",
    "inner_wall_line_width": "0.45",
    "inner_wall_speed": "300",
    "interface_shells": "0",
    "internal_bridge_support_thickness": "0.8",
    "internal_solid_infill_line_width": "0.42",
    "internal_solid_infill_speed": "250",
    "ironing_flow": "10%",
    "ironing_spacing": "0.15",
    "ironing_speed": "30",
    "ironing_type": "no ironing",
    "layer_height": "0.2",
    "line_width": "0.42",
    "max_bridge_length": "10",
    "max_travel_detour_distance": "0",
    "min_bead_width": "85%",
    "min_feature_size": "25%",
    "minimum_sparse_infill_area": "15",
    "name": "process",
    "only_one_wall_top": "1",
[... 91 lines cut ...]
debian@ns3017700:~/slice-bambu$ podman run -ti --rm -v /tmp/.X11-unix:/tmp/.X11-unix -v ~/slice-bambu/:/me:rw localhost/nathael/bambustudio /BambuStudio/build/package/bin/bambu-studio --debug 5    --curr-bed-type "Cool Plate" --load-settings "/me/machine.json;/me/process.json" --load-filaments "/me/filament.json" --export-settings /me/settings.json -slice 0 --export-3mf /me/cl-outputV5.3mf  /me/08.stl
[2023-06-28 07:28:48.750512] [0x00007f5d630a7940] [trace]   Initializing StaticPrintConfigs
[2023-06-28 07:28:48.762027] [0x00007f5d630a7940] [info]    start_gui=false

[2023-06-28 07:28:48.762122] [0x00007f5d630a7940] [info]    Will start to read model file now, file count :1

[2023-06-28 07:28:48.762141] [0x00007f5d630a7940] [info]    plate_to_slice=0, normative_check=1
[2023-06-28 07:28:48.762151] [0x00007f5d630a7940] [info]    read model file:/me/08.stl

[2023-06-28 07:28:48.889722] [0x00007f5d630a7940] [debug]   TriangleMesh::repair() started
[2023-06-28 07:28:48.899050] [0x00007f5d630a7940] [debug]   TriangleMesh::repair() finished
[2023-06-28 07:28:48.921367] [0x00007f5d630a7940] [info]    object 08.stl, id :5, from stl or other 3mf

[2023-06-28 07:28:48.921436] [0x00007f5d630a7940] [info]    run:before load settings, file count=2

[2023-06-28 07:28:48.921454] [0x00007f5d630a7940] [info]    operator():load setting file /me/machine.json, with rule 1

[2023-06-28 07:28:48.921715] [0x00007f5d630a7940] [info]    no substitutions performed from file /me/machine.json

[2023-06-28 07:28:48.921732] [0x00007f5d630a7940] [info]    operator():load setting file /me/process.json, with rule 1

operator():file /me/process.json's from  unsupportedrun found error, return -5, exit...
[2023-06-28 07:28:48.922209] [0x00007f5d630a7940] [info]    cli_callback_mgr_t::stop enter.
[2023-06-28 07:28:48.922215] [0x00007f5d630a7940] [info]    cli_callback_mgr_t::stop not started before, return directly.
```

lanewei120, 2023-06-28:
[presets_for_testing.zip](https://github.com/bambulab/BambuStudio/files/11892078/presets_for_testing.zip)
I have checked the logic 
and confirmed currently only system preset is supported to passed to CLI

I just attached the above full system presets for your test
please have a try

lanewei120, 2023-06-28:
just found a bug which will cause this issue, and we will fix it soon,

if you are building codes by yourself
you can apply this patch to your codes and have a try

diff --git a/src/BambuStudio.cpp b/src/BambuStudio.cpp
index a65c31eb1..2d75185a1 100644
--- a/src/BambuStudio.cpp
+++ b/src/BambuStudio.cpp
@@ -1173,7 +1173,7 @@ int CLI::run(int argc, char **argv)

         std::set<std::string> different_keys_set(different_keys.begin(), different_keys.end());
         BOOST_LOG_TRIVIAL(info) << boost::format("update process config to newest, different size %1%")%different_keys_set.size();
-        int ret = update_full_config(m_print_config, load_machine_config, different_keys_set);
+        int ret = update_full_config(m_print_config, load_process_config, different_keys_set);
         if (ret)
             flush_and_exit(ret);
     }

NathaelB, 2023-06-28:
Perfect, i have a good settings.json in out !

Thanks ! 🙏

closed 2023-06-29 by lanewei120

SomeoneWeird, 2023-07-27:
> I have checked the logic and confirmed currently only system preset is supported to passed to CLI

@lanewei120 Sorry for bumping an old issue, but is there plans to change this?

lanewei120, 2023-07-27:
> > I have checked the logic and confirmed currently only system preset is supported to passed to CLI
> 
> @lanewei120 Sorry for bumping an old issue, but is there plans to change this?

I am afraid currently we don't have such plan

SomeoneWeird, 2023-07-27:
> > > I have checked the logic and confirmed currently only system preset is supported to passed to CLI
> > 
> > 
> > @lanewei120 Sorry for bumping an old issue, but is there plans to change this?
> 
> I am afraid currently we don't have such plan

Would a PR be accepted if I did the work?

lanewei120, 2023-07-27:
we had a short discuss just now,
we will support to pass user profile later,
however, the user profile which including full values need to extracted before passed to CLI

SomeoneWeird, 2023-07-27:
> we had a short discuss just now, we will support to pass user profile later, however, the user profile which including full values need to extracted before passed to CLI

Understood, this is completely acceptable for my use-case. I know it might be silly to ask, but is there a priority/time estimate for this work? I might try and change some stuff locally to get it working in the meantime, if it's going to take a while.

arirangz, 2025-04-30:
Hello,

I am using the following proccess.json:
`{
     "type": "process",
    "brim_type": "outer_only",
    "enable_support": "1",
    "from": "User",
    "inherits": "0.20mm Standard @BBL A1M",
    "initial_layer_infill_speed": "100",
    "initial_layer_speed": "30",
    "name": "0.20mm Standard @BBL A1M - PeTG Round shape",
    "print_settings_id": "0.20mm Standard @BBL A1M - PeTG Round shape",
    "skirt_loops": "1",
    "sparse_infill_density": "75%",
    "top_surface_pattern": "concentric",
    "version": "1.8.0.25",
    "wall_loops": "7",
		"compatible_printers": [
        "Bambu Lab A1 mini 0.4 nozzle"
    ]
}
`

When I open the generated 3MF, the sparse_infill_density is not 75% but the default 15% value. So the overridden value are not being taken.

How can I fix this?

Someone said that we have to export the full settings but when I export the process, I am only getting the short version with inherits.

Thanks

arirangz, 2025-04-30:
@lanewei120 is it fixed?
@NathaelB @SomeoneWeird  Did you make it working?

thanks