bambulab/BambuStudio #1966: Top and bottom layer pattern - left-to-right continuous

OrdduShmoo, 2023-06-28:
None of the patterns for top/bottom layer currently take into account multi-colour filament.  Because they jump around and/or leave gaps to be filled later in the layer, they can leave obvious lines when the filament isn't run contiguously.  Even worse on gradient filament where there can be a jump in colour on a flat plane.  See photo.

Would be solved by a left-to-right top/bottom layer patteen which just jumps over any spaces in the layer, but goes from corner to corner regardless.  This may be slow and inefficient in the grand scheme of things, but only for the very top and/or bottom layer.  The other current top/bottom patterns each have their own version of the issue, and many are too ornate when you just want a straight, uniform appearance.

Tweaked various infill settings as described on Bambu FB, but nothing stops the head jumping about in sections.

Pictures: the problem in a real print; the current head movement pattern; the suggested movement pattern to add.
[image]
[image]
[image]

SaltWei, 2023-06-28:
之前提过一样的方案，不过怀疑可能会因为大量travel增加漏料，降低打印质量。
@QingZhangBambu 实现一下，做实验看看

QingZhangBambu, 2023-07-03:
#1869

SaltWei, 2023-07-03:
> #1869

不是这个单子，是内部提的。提的时候你还没毕业