bambulab/BambuStudio #1967: Wrong number of layers on MacOS version

mctazman, 2023-06-28:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 13.4.1

### Additional system information

[image]

### Printer

X1carbon/AMS

### How to reproduce

[tail.3mf.zip](https://github.com/bambulab/BambuStudio/files/11893716/tail.3mf.zip)
Slice the 3mf file on MacOS and Windows

### Actual results

[image]
As you can see in this screenshot, the left (MacOS version) shows the wrong number of layers for the print, while the right (WIndows 11 VM, Windows version) shows the correct number of layers

### Expected results

The correct amount of layers post-slice on Mac and Windows.

### Project file & Debug log uploads

[tail.3mf.zip](https://github.com/bambulab/BambuStudio/files/11893746/tail.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-06-29:
@mctazman 
Hello, thanks for reporting this issue

as we don't have the same mac version with you,
we have tried the same 3mf with 1.6.2.4 on following mac versions:
Mac 13.2.1
Mac 12.2.1
both of them can not reproduce this issue

it seems the support's height is not aligned with the normal print
so the support's layers will be treated as different layers
to confirm this issue
please turn off the support, to check the layer count again

we doubt it is related with computing precision
does other model have this issue?

SaltWei, 2023-06-29:
> we doubt it is related with computing precision

Yes, we met other issues on specific macos system which is caused by float computing precision. Finally wrong cooling speed was applied.

We doubt this issue is caused by same reason as well but still can't reproduced at the moment.

mctazman, 2023-06-30:
@lanewei120 The fault doesn't happen on all models. See, for instance, attached 3mf file. There's no error in layer height when slicing.
[brooms.3mf.zip](https://github.com/bambulab/BambuStudio/files/11914011/brooms.3mf.zip)
It could be an Apple Silicon issue, but I doubt it, as issue #1969 also displays weird layer number behaviour, but then on a WIndows machine.

lanewei120, 2023-06-30:
> @lanewei120 The fault doesn't happen on all models. See, for instance, attached 3mf file. There's no error in layer height when slicing. [...]

very strange, still can not reproduce issue 1969 also
we will try to find more machines to have a try

for this issue, if turn off support ,is this issue still there?

mctazman, 2023-06-30:
I've done 3 tests:

support off / prime tower on -> correct number of layers (607)
support on / prime tower off -> incorrect number of layers (625)
support off / prime tower off -> correct number of layers (607)
[image]
[image]
[image]

So, prime tower's a factor as well, apparently

SaltWei, 2023-06-30:
> So, prime tower's a factor as well, apparently

This is normal. Primer tower only works when all extrusion(objects and support) has same layer height.
When primer tower is disabled, the support can generate independent layer height with model, so the layer counter is larger.
Layer height of support is aligned with object when primer tower is enabled. So the later number may be lower.

mctazman, 2023-06-30:
> > So, prime tower's a factor as well, apparently
> 
> This is normal. Primer tower only works when all extrusion(objects and support) has same layer height. When primer tower is disabled, the support can generate independent layer height with model, so the layer counter is larger. Layer height of support is aligned with object when primer tower is enabled. So the layer number is lower.

I understand, but my point is that both turned on, the layer count goes from 607 (both off) to 627 (only supports) to 914 (both on). So the prime tower has a significant impact, whilst it's obviously not adding layers, just extruded plastic.

SaltWei, 2023-06-30:
> 914 (both on)

Can not reproduce at the moment. We are keeping trying. My version is 1.6.2.4, windows 10 system.
[image]

Hologos, 2023-07-03:
> > 914 (both on)
> 
> 
> 
> Can not reproduce at the moment. We are keeping trying. My version is 1.6.2.4, windows 10 system.
> 
> [image]
> 
> 

Hello, guys. From the image, OP is using 0.08mm preset but from the name, it's his own. @SaltWei is using stock 0.08mm preset. You should use his profile from included 3mf file.

BambooBugFixer, 2023-07-13:
I used a Mac with Ventura 13.4.1 and a bamboo studio with a version of 1.6.2.4. I sliced your model with the support and prime tower on and the number of layers is right.  
[image]

Only when I turn the support on and the prime tower off, there are a little bit more layers. But this is normal.
[image]

mctazman, 2023-08-18:
> @mctazman<https://github.com/mctazman> Hello. Does this issue still exist on your environment with 1.7.4 version?

I’ll have to try later. Am currently printing another project. Will check later today.

mctazman, 2023-08-20:
OK, tad later than I wanted to, but finally got to test the layer-issue. I gave it the worst possible case I could quickly think up & seemingly it works perfectly:
[image]
[image]
If wanted, I can upload the 3mf file, but it seems that the issue's solved.

SaltWei, 2023-08-21:
> but it seems that the issue's solved.

Thank you very much. So we can close this issue now.
Please reopen if could reproduce again.

closed 2023-08-21 by SaltWei