bambulab/BambuStudio #1969: Displayed number of layer count jumping

Panda631, 2023-06-30:
### Bambu Studio Version

1.6.4.2

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Additional system information

Ryzen7

### Printer

X1C & 2xAMS

### How to reproduce

1. open attached file
2. Slice the plate.
3. Move the right slider to the most top position. 
4. Layer count is 611
5. Move the slider below layer 590.
6. Layer count jumps to 1147!

[image]

[image]

### Actual results

Layer count jumps

### Expected results

Layer count should be correct

### Project file & Debug log uploads

[Layer_counter.zip](https://github.com/bambulab/BambuStudio/files/11913592/Layer_counter.zip)

[debug_Fri_Jun_30_07_58_12_11356.log.zip](https://github.com/bambulab/BambuStudio/files/11913658/debug_Fri_Jun_30_07_58_12_11356.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

redsector72, 2023-09-01:
This bus still present, fixed in Orca

lanewei120, 2023-09-14:
@redsector72 @Panda631 

previously we thought it is a same issue with 1967
and we can not reproduce it here 
[image]
[image]

are you using the 3mf attached above?
and are you using the newest version of Bambu Studio(v1.7.4)?

Panda631, 2023-09-15:
This issue seems to be solved in V1.7.4.52.
I can't reproduce any more.

SaltWei, 2023-10-07:
> This issue seems to be solved in V1.7.4.52.

Thanks for checking!

But actually we do nothing.
Close at the moment. Reopen if can be reproduced again.

closed 2023-10-07 by SaltWei