bambulab/BambuStudio #1971: Reduce filament waste at the beginning and when swapping filament with AMS

capsel22, 2023-07-01:
**Is your feature request related to a problem? Please describe.**
Would it be possible to reduce the amount the printer purgers filament at the beginning of the print and when changing filament with AMS?
Last color print I did I had nearly 1kg of waste.
I agree we need the purge, but the current amount seem high

**Describe the solution you'd like**
Reduce purge by third or half. Or allow us to specify manually.

**Describe alternatives you've considered**
N/A

**Additional context**
Add any other context or screenshots about the feature request here.

DokuKaefer, 2023-07-01:
You can setup the flushing volumes already
[image]

DokuKaefer, 2023-07-01:
And for the waste at the beginning, you can edit the start code. Search for the following block and the two G1 parameters.
The E50 is the length in mm.

M109 S250 ;set nozzle to common flush temp
M106 P1 S0
G92 E0
_**G1 E50 F200**_
M400
M104 S[nozzle_temperature_initial_layer]
G92 E0
_**G1 E50 F200**_
M400
M106 P1 S255
G92 E0
G1 E5 F300
M109 S{nozzle_temperature_initial_layer[initial_no_support_extruder]-20} ; drop nozzle temp, make filament shink a bit
G92 E0
G1 E-0.5 F300

closed 2023-07-01 by capsel22

Yegorich555, 2025-04-07:
Option `Flushing volumes` affects only on changing filament, but not for starting

Is there any way to setup `waste at the beginning` without direct changing G-code?
Because it's not convenient to do it on each printing