bambulab/BambuStudio #1972: Software filament colour selection not responding

Fossilix, 2023-07-03:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Additional system information

Appli M1
8GB

### Printer

Bambu Lab P1P

### How to reproduce

Device AMS A1/A2/A3/A4 filament colour selection Edit
Select Orange/ red/ blue / dark blue - colours not responding to selection. Other colour are ok.

### Actual results

Unable to select the right colour filament for the AMS spools

### Expected results

The selected colour for the filament must show in the AMS filament channel

### Project file & Debug log uploads

Unable to log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Fossilix, 2023-07-03:
In the updated Bambu Lab studio for MAC OS, I am not able to select the right colour for the filament in the AMS and the side load. Some of the colours get selected, but some colours not responding to colour change when selected. For example if I select orange, red, blue, and dark blue on the left side bottom colours, they are not responding to the selection.
May be a bug in the software update.

walterwongbbl, 2023-07-18:
@Fossilix  Hi. Could you please test the lastest version? The latest version is 1.7.0.65.

walterwongbbl, 2023-08-24:
@walterwongbbl

closed 2023-09-07 by walterwongbbl