bambulab/BambuStudio #1973: First layer heights for X1C with 0.2 nozzle seem incorrect

sprior, 2023-07-04:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Select Bambu Lab X1 Carbon 0.2 nozzle
2. Look at the Layer Height while changing the standard process settings:

### Actual results

With "0.06mm Standard @BBL X1C 0.2 nozzle" preset the layer height is 0.06 and the first layer height is 0.1 - seems reasonable.
0.08mm preset looks good too with a 0.1 first layer height.

### Expected results

But with 0.10, 0.12, 0.14 presets the first layer height stays at 0.1mm - I haven't seen a first layer height that's smaller than the regular height.  Should each of these be adjusted?

### Project file & Debug log uploads

[untitled.zip](https://github.com/bambulab/BambuStudio/files/11943639/untitled.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-07-06:
> But with 0.10, 0.12, 0.14 presets the first layer height stays at 0.1mm - I haven't seen a first layer height that's smaller than the regular height. Should each of these be adjusted?

The height of the first layer should not be too low, but at the same time, it should not be too high. A first layer height that is too high means that the width-to-height ratio of the extruded filament is relatively small, which is not beneficial for adhesion of the first layer. Therefore, the first layer height of Bambu is half the diameter of the nozzle, which is in line with expectations.

SaltWei, 2023-07-11:
Not a bug. The initial layer height is as our expectation. If you met some other issue about slicer, don't hesitate to raise tickets. 
Thank you very much!

closed 2023-07-11 by SaltWei