bambulab/BambuStudio #1974: Filament calibration dialog is empty and can't set

michelem09, 2023-07-05:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 13.4.1

### Printer

Bambu lab P1P

### How to reproduce

1. Go to device
2. Click "Edit" icon on Ext Pool image
3. Filament select is empty

### Actual results

I can't select any filament

[image]

### Expected results

I expect to be able to select a filament

### Project file & Debug log uploads

It's not related to a specific project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

michelem09, 2023-07-14:
Please let me know me if I can help fixing this I'm available for debugging. The 1.7.0 beta does the same.

michelem09, 2023-07-14:
I just downloaded the 1.7.0 beta2 and it seems to be fixed! Great!
I will do some tests.

[image]

walterwongbbl, 2023-07-18:
it's great.
We are also testing calibration

closed 2023-07-18 by walterwongbbl