bambulab/BambuStudio #1975: Color/Paint Options

TheAtomicSoul, 2023-07-05:
Select all painted areas of a color (So you can only paint there, and invert it so it avoids it.)
Lock already painted color areas so it cannot be changed. 

Change color order (Auto renumber them)

Haidiye00, 2023-10-24:
Can the requirements be described in more detail? If feasible, we will consider it

TheAtomicSoul, 2023-10-25:
> ents be described in more detail? If feasible,

Essentially a mask option, so you can: 
Select an area, then you can only paint in that area. 
Select an area, then invert selection, so you can avoid that area. 
Lock a color, so you cannot change the painted areas of that color (prevent accidentally painting areas.)

Haidiye00, 2024-05-09:
May I ask if you can use pictures or similar diagrams to describe your ideas

TheAtomicSoul, 2024-05-09:
> May I ask if you can use pictures or similar diagrams to describe your ideas

It's basically a selection mask. (didn't think of the right word when I wrote the post.)
Like in Photoshop: you click the layer and it selects everything in the layer. You click the option to mask the empty areas of the layer so then only whats already painted on the layer is "editable".

Haidiye00, 2024-05-10:
Sorry, we may not be able to do this at the moment and still don't understand the benefits of doing so. It would be even better if you could use a diagram to describe it.

TheAtomicSoul, 2024-05-10:
>... still don't understand the benefits of doing so. It would be even better if you could use a diagram to describe it.
Not sure how to make an image that'd visually show the difference without tons of photos, so while I think of that, here's a few scenarios. 

1: Lock Color: (Separate option for each color) Prevents you from painting over this color with another color. This would allow you to paint without worry of messing up the work done on these colors. Helpful because the "fill" tool isn't perfect. 
2: Select Color: Ensures you ONLY paint on areas that were already painted on with that specific color. Allows you to paint while limiting it to replacing only the color selected. (Does not work on Locked Color layers, which that colors lock option would have to be off)
3: Select Clear: Limits your painting area to places that have not already been manually painted on.

closed 2024-05-15 by Haidiye00