bambulab/BambuStudio #1978: Ability to set Brim Material

retoid, 2023-07-05:
**Is your feature request related to a problem? Please describe.**
Currently there is no way to set which material is used for the Brim. When printing multi color with the AMS this causes an issue of having to clean up the brim a lot more since it's a different color. Having the option to set Brim Material would make life a lot easier and allow clean up of brims much simpler and quicker.

**Describe the solution you'd like**
Allow the option to select Brim Material.

**Describe alternatives you've considered**
As an alternative I have set the Brim-object Gap larger, however this still doesn't solve the issue. If not big enough, the material bonds to the main object, if set too big then the brim no longer serves it's purpose.

**Additional context**

schmitzna, 2023-08-22:
This happens all the time when I print a graphic on the first layer.  The brim is always the graphic filament, not the body of the part that touches the brim.  It can be very difficult to remove the offending color brim

RangerCD, 2023-12-22:
You can achieve this by customizing `First layer filament sequence`, under `Objects` tab.
https://forum.bambulab.com/t/ability-to-choose-different-ams-slot-for-brim/3776/12?u=rangercd

petervanstark, 2024-07-29:
Answer is not valid: Approach only allows to select which color used in part will be used first...

I want to use PVA as brim.

Ok, I can work around by adding dummy objects but it is not really an answer.

Please add selector for brim to allow us to select support material.